mantidproject/mantid #36701: Manual Testing ISIS SANS

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

thomashampson, 2024-01-25:
It seems like steps 5 and 10 of [Runs table editing](https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html#id5) contradcit each other?

thomashampson, 2024-01-25:
Remove reference to detector combo box here https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html#id6

thomashampson, 2024-01-25:
Step 6 of https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html#id8 might need updating. It saves more files than it says it should.

thomashampson, 2024-01-25:
`Switch to 2D and manually untick CanSAS (since we have manually changed the save options at this point)`
CanSAS is disabled at this point so we can't untick it

cailafinn, 2024-01-25:
Not a regression. Caused when trying to colourfill a HAB 2D reduction:
```
Traceback (most recent call last):
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/workbench/plugins/workspacewidget.py", line 375, in _action_double_click_workspace
    TableWorkspaceDisplay.supports(ws)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/mantidqt/widgets/workspacedisplay/table/presenter.py", line 136, in supports
    return TableWorkspaceDisplayModel.supports(ws)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/mantidqt/widgets/workspacedisplay/table/model.py", line 55, in supports
    raise ValueError("The workspace type is not supported: {0}".format(ws))
ValueError: The workspace type is not supported: first_time_HAB_2D_2.2_10.0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/cbook/__init__.py", line 307, in process
    func(*args, **kwargs)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colorbar.py", line 495, in update_normal
    self._draw_all()
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colorbar.py", line 530, in _draw_all
    self._process_values()
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colorbar.py", line 1103, in _process_values
    b = self.norm.inverse(b)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colors.py", line 1707, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
Traceback (most recent call last):
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/workbench/plugins/workspacewidget.py", line 375, in _action_double_click_workspace
    TableWorkspaceDisplay.supports(ws)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/mantidqt/widgets/workspacedisplay/table/presenter.py", line 136, in supports
    return TableWorkspaceDisplayModel.supports(ws)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/mantidqt/widgets/workspacedisplay/table/model.py", line 55, in supports
    raise ValueError("The workspace type is not supported: {0}".format(ws))
ValueError: The workspace type is not supported: first_time_HAB_2D_2.2_10.0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/cbook/__init__.py", line 307, in process
    func(*args, **kwargs)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colorbar.py", line 495, in update_normal
    self._draw_all()
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colorbar.py", line 530, in _draw_all
    self._process_values()
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colorbar.py", line 1103, in _process_values
    b = self.norm.inverse(b)
  File "/Users/caila.finn/mambaforge/envs/mantid-env/lib/python3.10/site-packages/matplotlib/colors.py", line 1707, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
```

thomashampson, 2024-01-25:
`main-detector` is no longer an option when following https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html#id9

thomashampson, 2024-01-25:
`Check the values in the first two text boxes at the top (Centre Position - Rear) have changed when it has finished running.` only the first value changed

thomashampson, 2024-01-25:
no tooltips for wavelength text boxes in `Settings>Q, Wavelength, Detector Limits` tab. But they weren't there in 6.8 either.

thomashampson, 2024-01-25:
this manual testing took 1 hour and 7 minutes to complete.