mantidproject/mantid #36705: Manual Testing Project Recovery

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

cailafinn, 2024-01-25:
Start: 11:59

cailafinn, 2024-01-25:
When two instances are running. Project Recovery displays the correct checkpoint on the dialog, but the loading loads the last-but-one checkpoint. This can be seen in test 5 of the instructions.

thomashampson, 2024-01-25:
When I ran test 4, it recovered the workspaces from test 3 👎

adriazalvarez, 2024-01-25:
I'm also experiencing incorrect retrieval of previous workspaces in windows in test 5 and 6

thomashampson, 2024-01-25:
If it loads the previous recovery, then when I open an extra workbench it will ask again and open the correct one.

thomashampson, 2024-01-25:
We spent 50 minutes testing it

adriazalvarez, 2024-01-25:
I've opened the issue. #36737 . I have tried in v 6.8, it seems to be happening to.

closed 2024-01-25 by adriazalvarez