mantidproject/mantid #36710: Missing signal warning on Data Analysis Interface.

adriazalvarez, 2024-01-24:
**Describe the bug**

There is a QObject missing signal warning on the view of the Data Analysis interface, likely caused by the refactoring of the presenter. 

**To Reproduce**

1. Open workbench from conda. 
2. Open Data Analysis interface
3. Warning should be issued on the console. 
**Expected behavior**

The warning should not be there.
**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Windows10
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

Added to 6.10 milestone, as it is not a critical issue for next release.

robertapplin, 2024-01-29:
Looks like this is fixed by #36624

adriazalvarez, 2024-01-29:
Yes, probably. I'll closed it then once the other is merged.

sf1919, 2024-01-29:
I've changed the milestone as it will go into v6.9

sf1919, 2024-01-30:
Can this now be closed?

adriazalvarez, 2024-01-30:
Fixed by #36624

closed 2024-01-30 by adriazalvarez