mantidproject/mantid #36720: Manual Testing ORNL Powder Diffraction

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See http://www.mantidproject.org/PowderDiffractionReduction

Kvieta1990, 2024-02-07:
Powder diffraction suites at ORNL all tested manually with the latest version of nightly and they are all working fine.

Kvieta1990, 2024-02-07:
I should close this ticket. BTW, the tested routines include,

NOMAD: `mantidtotalscattering`

POWGEN: GUI within Mantid Workbench for data reduction

HB2A: Normal routine for data reduction - we tested it through the autoreduction pipeline which contains all the recipes needed for the HB2A data reduction with Mantid.

HB2C: On the powder side, we tested with the event filtering script which contains both the event filtering operations and the follow-up data reduction with `WANDPowderReduction` routine.

closed 2024-02-07 by Kvieta1990