mantidproject/mantid #36726: Unhandled exception when plotting live data from fake histogram instrument

rbauststfc, 2024-01-24:
**Describe the bug**

Found during manual testing for the 6.9 release. Trying to open a plot on a workspace that is being updated with data from the fake `ISIS_Histogram` instrument causes an unhandled exception. In Mantid version 6.8 this would log an error message but not trigger the unhandled exception, so it appears to be a regression.

**To Reproduce**

Go through the following steps from the live data manual testing instructions to get the live data running:

1) https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html#set-up-the-test-facility
1) https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html#isis-histogram
1) https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html#starting-a-live-data-session (just steps 1 and 2 here)
1) https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html#live-histogram-data

Then, with live data running, right click on the workspace in the ADS and try to open a plot (select a single spectrum to plot). This should cause an unhandled exception.

**Expected behavior**

In previous versions an error was logged but it didn't cause the unhandled exception.

**Platform/Version (please complete the following information):**
 - Mantid Version - introduced since version 6.8.

thomashampson, 2024-01-31:
For me this is throwing an unhandled exception in 6.8

GuiMacielPereira, 2024-02-01:
Unassigning this from myself since I could not reproduce this issue on Linux (Pop_OS! 22.04).

PR #37683 (2024-07-25): Fix error when plotting live data
Instead of passing the workspaces references, pass the names of the workspaces to the plot function. With live data, after selecting which spectra to plot in the pop-up dialog, the workspace can be overwritten, at which point the reference is going to be invalid.

Fixes #36726

### To test:
See #36726 for details on how to reproduce.

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used as part of the gatekeeper process, so please comment c [... cut]
closed 2024-07-25 by SilkeSchomann