mantidproject/mantid #36731: Fix workbench testing documentation

thomashampson, 2024-01-25:
The information about this issue is incorrect. We no longer have groups 1-4 and the documentation that is linked also talked about grouped tests and the documentation not being up to date after v6.2

_Originally posted by @sf1919 in https://github.com/mantidproject/mantid/issues/36694#issuecomment-1907864454_

perenon, 2024-01-25:
In [this page](https://developer.mantidproject.org/Testing/Core/Core.html) (referenced in the main documentation) the link to "3D and volumetric data" goes directly to [matplotlib documentation](https://matplotlib.org/stable/plot_types/3D/index.html#d-plots)

Also, the data used in [this section](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/plotting/01_basic_plot_scripting.html#d-surface-and-wireframe-plots) `PG3_733.nxs` does not appear in the standard Mantid Training Data

jclarkeSTFC, 2024-05-20:
The issue name has been changed from "Groups 1-4" to "Core" in this PR (https://github.com/mantidproject/documents/pull/123), not yet merged at the time of writing.

The link that goes to the matplotlib documentation, that seems okay to me, is there somewhere in the Mantid docs that is more suitable?

The `PG3_733.nxs` file is in the Usage Examples folder, I could add a line to the docs saying where it is, unless all of the files used for these examples are supposed to be in a particular folder? In that case I cold just change which file is used for that plot to one that's in the SampleData-ISIS folder.

PR #37487 (2024-06-12): Add note about where to find plotting example file
Fixes #36731

### To test:

Check the docs build okay.

*This does not require release notes* because **it's a trivial change to the docs**

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used as part of the gatekeeper process, so please comment clearly on what you have checked during your review.** If changes are made to the PR during the review process then your final comment will be the most important for gatekeepers. I [... cut]
closed 2024-06-12 by SilkeSchomann