mantidproject/mantid #36744: Missing background workspace in manual test

jclarkeSTFC, 2024-01-26:
Following the engineering diffraction manual test instructions (https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html), when doing test 6, point 1, loading a run does not automatically generate the background workspace with suffix `_bg`.

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 10
 - Mantid Version 6.8.20240123.1153

jclarkeSTFC, 2024-01-26:
This is in 6.8

warunawickramasingha, 2025-02-17:
I think the updated test instructions have captured this correctly and now this particular test is [test 7](https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html#id11 ) and the interested suffix is `_bgsub`.

RichardWaiteSTFC, 2025-02-17:
Thanks @warunawickramasingha for checking this - I agree test 6 and 7 both have correct workspace suffix