mantidproject/mantid #36745: Engineering diffraction window disappears

jclarkeSTFC, 2024-01-26:
Following the manual testing instructions (https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html), when doing test 7, point 6, performing a fit through Fit > Fit closed the Eng Diff tab. When the tab was opened again, the fit had been successful and no data was deleted.

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 10
 - Mantid Version 6.8.20240123.1153

jclarkeSTFC, 2024-01-26:
I got this to happen in 6.8

jclarkeSTFC, 2024-05-03:
I've just tried this in 6.9 and couldn't reproduce it

warunawickramasingha, 2024-05-03:
Thanks James for updating. I also tried to re-create this in 6.9 but couldn't. Hence closing the issue.

closed 2024-05-03 by warunawickramasingha