mantidproject/mantid #36747: GSAS II crash

jclarkeSTFC, 2024-01-26:
Following the manual testing instructions (https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html), when doing test 10, point 8, with the range 20000-45000, Mantid crashed due to a missing file error. The first run we did with GSAS II worked fine, but after changing the limits it didn't work again.

**Platform/Version (please complete the following information):**
 - OS: IDAaaS Rocky
 - OS Version: 8?
 - Mantid Version 6.8.20240123.1153

warunawickramasingha, 2024-01-26:
This issue can be re-created on Mantid-6.6, 6.7 and 6.8 as well on IDAaaS. It occurs since GSAS-2 tool is not creating `<project_name>_cell_parameters_Fe_gamma.txt` file for the provided TOF range.

warunawickramasingha, 2024-01-30:
#36772 was raised to investigate why the reflection files are not generated for some TOF ranges only on IDAaaS environments in comparison to the other OS such as Windows, and Linux which successfully created the output files for the same TOF range.

warunawickramasingha, 2024-01-31:
Manually closing the issue since the fix is already merged and the GitHub action for auto-closing has not closed.

PR #36767 (2024-01-30): 36747 avoid crash on GSAS-II
#### Summary of work
GSAS-II tab crashed on IDAaaS environment because of the unavailability of files expected to be generated from GSASIIscriptable interface. The invocation of the interface happens in a subprocess and the code has not checked the return codes of the call which contained errors with the reasons. By handling the error codes it can be decided whether to proceed further and avoid the crash.

Note: Although the crash is avioded with the fix, it needs to further investigate why GSASIIscriptable is not generating the expected output files(the reflection [... cut]
closed 2024-02-01 by sf1919