mantidproject/mantid #36755: Filtering weighted events stopped working

AndreiSavici, 2024-01-26:
**Original reporter:** @granrothge SNS

**Describe the bug**
GenerateEventFilter/FilterEvents stopped working for weighted events

**To Reproduce**
Use systemtest datafile, and run 
```python
from mantid.simpleapi import *

raw=Load('SEQ_176472.nxs.h5')
GenerateEventsFilter(InputWorkspace=raw, OutputWorkspace='evf',InformationWorkspace='evf_info',
                     StartTime="0",StopTime="600", TimeInterval=200)
FilterEvents(InputWorkspace=raw, SplitterWorkspace='evf', OutputWorkspaceBaseName='raw_split', 
             InformationWorkspace='evf_info', FilterByPulseTime=True, GroupWorkspaces=True)

normalized = NormaliseByCurrent(InputWorkspace=raw)
GenerateEventsFilter(InputWorkspace=normalized, OutputWorkspace='evfn',InformationWorkspace='evfn_info',
                     StartTime="0",StopTime="600", TimeInterval=200)
FilterEvents(InputWorkspace=normalized, SplitterWorkspace='evfn', OutputWorkspaceBaseName='norm_split', 
             InformationWorkspace='evfn_info', FilterByPulseTime=True, GroupWorkspaces=True)

print('raw\tnorm')
for i in range(3):
    print(f"{mtd['raw_split'][i].getNumberEvents()}\t{mtd['norm_split'][i].getNumberEvents()}")
```

**Expected behavior**
The number of events in each filtered workspace should be the same for normalized as for raw. It works in the release version, it does not in nightly

**Platform/Version (please complete the following information):**
 - OS: RHEL, Ubuntu
 - Mantid Version 6.8 vs nightly

granrothge, 2024-01-29:
If this helps it was first noticed in the nightly builds on December 5, 2023.

peterfpeterson, 2024-01-30:
From the first `GenerateEventsFilter` (combining tables)
```
start                                 stop                                 workspacegroup  title
930041879032615666	930042079032615666	0                          Time.Interval.From.2019-Jun-22 08:57:59.032615666.to.2019-Jun-22 09:01:19.032615666
930042079032615666	930042279032615666	1                          Time.Interval.From.2019-Jun-22 09:01:19.032615666.to.2019-Jun-22 09:04:39.032615666
930042279032615666	930042479032615666	2                           Time.Interval.From.2019-Jun-22 09:04:39.032615666.to.2019-Jun-22 09:07:59.032615666
```
which is 200s steps as requested

`GenerateEventsFilter` on the normalized workspace gives
```
start                                 stop                                 workspacegroup title
930041879032615666	930042079032615666	0                         Time.Interval.From.2019-Jun-22 08:57:59.032615666.to.2019-Jun-22 09:01:19.032615666
930042079032615666	930042279032615666	1                         Time.Interval.From.2019-Jun-22 09:01:19.032615666.to.2019-Jun-22 09:04:39.032615666
930042279032615666	930042479032615666	2                         Time.Interval.From.2019-Jun-22 09:04:39.032615666.to.2019-Jun-22 09:07:59.032615666
```
which also has 200s steps as requested. 

`CompareWorkspaces` also agrees that the generated filtering tables are identical.

peterfpeterson, 2024-01-30:
Adding more diagnostics reveals that the output workspaces from both the `TOF` and `WEIGHTED` input workspaces outputs `TOF` workspaces.

peterfpeterson, 2024-01-30:
The issue is that the OutputWorkspace does not get the correct type during `FilterEvents::createOutputWorkspaces()`. When `TimeSplitter::splitEventList` is run, it calls `EventList::addEventsQuickly()` which appends to the correct concrete `std::vector` based on template type. In its current state, this means that the events get added to the `std::vector<WeightedEvents>`, but the `EventList` thinks its type is `TOF` so the events are hidden from all calls.

I perceive a couple of solutions for the issue which is not mutually exclusive:
1. Change `FilterEvents::createOutputWorkspaces()` to call `EventWorkspace::switchEventType()` for the donor workspace so each output gets the correct type
2. Change `TimeSplitter::splitEventList()` to use `EventList::switchTo()` out the output `EventList`s
3. Check the types and throw an exception if they don't match

Either way, the first task should be to convert the script in the description into a unit test. The unit test doesn't need to run normalize and can call `EventWorkspace::switchEventType()`  directly. The key to the test is that it compares the number of events and integrated intensity for `TOF` and `WEIGHTED` workspaces between the two cases.

peterfpeterson, 2024-01-31:
After a full night's sleep, Another solution is to correct the issue by overriding `ISpectrum::copyInfoFrom()` which does not have a specialization in  `EventList`. The specialization could call the parent method then set the `API::EventType`.

closed 2024-02-01 by github-actions[bot]