mantidproject/mantid #36756: Access Violation when opening and using multiple AddWorkspaceDialog's

robertapplin, 2024-01-29:
**Describe the bug**
There is a bug when opening multiple `AddWorkspaceDialog's from the Inelastic Data Analysis interface, Data Manipulations interface and the Fit Script Generator interface. If you close the second dialog you open and then click `Add` on the first dialog, there is an access violation because the raw pointer for the dialog is now invalid.

**To Reproduce**

1. Open Inelastic Data Analysis interface
2. On any of the tabs, click `Add Workspace`
3. Click `Add Workspace` again so that you now have two dialogs open
4. Close the second dialog you opened
5. Click `Add` on the first dialog you opened
6. Access violation!

**Expected behavior**
It should be possible to open multiple dialogs without an access violation occuring

**Platform/Version (please complete the following information):**
 - OS: All

sf1919, 2024-01-30:
Can this now be closed?

robertapplin, 2024-01-30:
Fixed by https://github.com/mantidproject/mantid/pull/36624

PR #36624 (2024-01-30): IAddWorkspaceDialog moved to a common location and no longer a qobject
### Description of work
This PR moves the `IAddWorkspaceDialog` class to the MantidQtWidgetsCommon library, and also combines the functionality of the `AddWorkspaceDialog` class with the `IndirectAddWorkspaceDialog` class because they are essentially the same thing. This has meant I could remove the `IndirectAddWorkspaceDialog` class and just use the standard `AddWorkspaceDialog` class from the MantidQtWidgetsCommon library.

This PR also makes sure the `IAddWorkspaceDialog` class is no longer a QObject.

Fixes #36565
Fixes [... cut]
closed 2024-01-30 by robertapplin