mantidproject/mantid #36771: Setting the title of an open plot doesn't update the plot

jclarkeSTFC, 2024-01-30:
**Describe the bug**
Setting the title or the axes labels from a script of a plot window that's already open won't update the plot. The window title (only relevant for the title, not the axes labels) will update, but not the actual title in the plot.

**To Reproduce**
Test script B from this page - https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/solutions/03_pim_sol.html#pim-sol

**Expected behavior**
The title of the plot should be `Oscillations` instead of `ws`

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 10
 - Mantid Version 6.9

GuiMacielPereira, 2024-01-31:
I tried to replicate this issue on Linux (Pop_OS! 22.04) but could not get this bug. The title of the plot updated as expected.

I had a look at the call that updates the title:
https://github.com/mantidproject/mantid/blob/1a4e46e868698c157b5eeb56d059c31824e9b1e6/qt/applications/workbench/workbench/plotting/figureinteraction.py#L287

and it's similar to how the x and y axis are updated, so it's worth checking if this bug is also happening when updating the x and y axis.

jhaigh0, 2024-02-01:
Problem is also on idaaas nightly version

closed 2024-02-14 by github-actions[bot]