mantidproject/mantid #36805: Change dQ/Q unit in ISIS Reflectometry ORSO file

rbauststfc, 2024-02-05:
In the ORSO file output by `SaveISISReflectometryORSO` we set the `value_is` value of error column `Qz` (resolution or dQ/Q) to `sigma`. However this is not the correct description of the resolution distribution that we calculate at ISIS and so it needs to be changed. Our scientists are still confirming what we should change it to.

This will be a short-term fix to ensure the file we provide in version 6.9 is accurate. Any longer term solutions will be tackled under a separate issue.

sf1919, 2024-02-12:
@thomashampson I think this should have closed. Have we made the fixes for auto closing on `release-next`

SilkeSchomann, 2024-02-12:
@rbauststfc Can we close this issue now?

rbauststfc, 2024-02-12:
@SilkeSchomann, yes, this can definitely be closed. I'll do that now.

PR #36821 (2024-02-09): Leave Q resolution column unit blank in ISIS Reflectometry ORSO file header
### Description of work

#### Summary of work

Sets the unit for the Q resolution (`Qz` column) in the ORSO file to blank. We previously had this set to `sigma`, but this is not correct and our scientists need more time to determine what it should be. We agreed to change it to be blank for the 6.9 release.

Fixes #36805. 

### To test:

1) Go to `Interfaces -> Reflectometry -> ISIS Reflectometry`.
1) On the Runs tab, enter Investigation ID `1120015` and Cycle `11_3`. Click the Search button.
1) From the search results, [... cut]
closed 2024-02-12 by rbauststfc