mantidproject/mantid #36816: Rogue lines in colorfill plot

jclarkeSTFC, 2024-02-06:
**Original reporter:** James Lord

**Describe the bug**
Original email:
This seems to have been around for a version or three but is still there in the beta nightly 6.8.20240202.1625 (Windows):

Load a data set with a lot of zeros in it and a few high valued bins (e.g. HIFI run 185451, which is dark counts and with the DAE not working properly so some spectra are completely empty)

Plot – Colour Fill. There should be some coloured in pixels with most of the graph dark blue/black (zero). But there may be a yellow horizontal line for example (narrower than one spectrum along that axis).

Resize the plot window with the mouse, and the line may change colour, change to dotted/dashed, or be accompanied by more horizontal and vertical lines.

The effect doesn’t show up if I load “normal” muon data with non-zero counts in most bins.

James

**To Reproduce**
- Load `HIFI 185451`
- Open colorfill plot
- Drag bottom right corner around to resize, random horizontal and/or vertical lines will appear in the plot.

**Expected behavior**

- No random lines

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: Windows
 - OS Version: 10
 - Mantid Version 6.8 (but present in earlier versions)

GuiMacielPereira, 2024-03-19:
I couldn't find the reason for why this is happening, but I did notice that simply updating the plot settings makes these lines go away

jhaigh0, 2024-05-08:
If we can't firgure this out, should we make sure James Lord knowns about the workaround you've figured out?

GuiMacielPereira, 2024-05-09:
Yeah, I can contact him to let him know

jclarkeSTFC, 2024-09-12:
This happens with `matplotlib` 3.8 as well