mantidproject/mantid #36836: FilteredTimeSeriesProperty returns different time-average values

peterfpeterson, 2024-02-09:
This was discovered when updating [pyrs](https://github.com/neutrons/PyRS/) from mantid v6.5 to mantid v6.8. The fundamental issue is that for a `FloatFilteredTimeSeriesProperty` which we'll call `tsp`, it is found that
```python
tsp.getStatistics().time_mean != tsp.timeAverageValue()
```
where the value from `tsp.getStatistics().time_mean` is correct.

**Original reporter:** @fanchercm

**Describe the bug**
The following script was generated by looking at how pyrs uses the `scan_index` log to create temporary `FilteredTimeSeriesProperty`ies and get the time average value. Fixing the bug will make the following script work
```python
from mantid.simpleapi import LoadEventNexus
from mantid.kernel import BoolTimeSeriesProperty, FloatFilteredTimeSeriesProperty

nexus_file = 'HB2B_1017.nxs.h5'
hidra_ws = LoadEventNexus(Filename=nexus_file, OutputWorkspace='hidra_ws', NumberOfBins=1, MetaDataOnly=True)

scan_index = hidra_ws.run()['scan_index']
tth = hidra_ws.run()['2theta']

for t,v in zip(scan_index.times, scan_index.value):
    print(t,v)

for start, stop, exp in ((1,2, 69.99525), (3,4, 80.), (5,6, 97.50225)):
    filter =   BoolTimeSeriesProperty('filter')
    filter.addValue(scan_index.times[start], True)
    filter.addValue(scan_index.times[stop], False)

    filtered_tsp = FloatFilteredTimeSeriesProperty(tth, filter)
    stats = filtered_tsp.getStatistics()
    print('scan_index', scan_index.value[start], 'expected tth', exp)
    print('min/max:', stats.minimum, stats.maximum)
    print('$$$ stats', stats.time_mean, stats.time_standard_deviation)
    print('$$$ avg ', filtered_tsp.timeAverageValue())
    # Statistics is correct
    assert abs(stats.time_mean - filtered_tsp.timeAverageValue()) < 1E-5
    # time average should find the "correct value"
    assert abs(filtered_tsp.timeAverageValue() - exp) < 1E-5
```

**Platform/Version (please complete the following information):**
 - Mantid Version nightlies during v6.9.0 code freeze

SilkeSchomann, 2024-02-19:
@peterfpeterson Can we close this now?

peterfpeterson, 2024-02-19:
I thought it would auto-close when the PR got merged. I was mistaken.

PR #36848 (2024-02-16): Fix bug in averageValueInFilter
### Description of work

Fix bug in `TimeSeriesProperty::averageValueInFIlter` where, in certain cases, the wrong averaged value was returned. 

#### Summary of work

In the very specific test case that you have a `FilteredTimeSeriesProperty` object with a single value in the filter, `TimeSeriesProperty::averageValueInFilter` would return the first value in the object not the first item in the filter. This could cause the value to fall outside of the filter and provide incorrect data. 

Fixes #36836. 
Refers to [EWM 3950](https://ornlrse.clm.ibmcloud.com/ccm/web [... cut]