mantidproject/mantid #36852: Cannot export cuts from 3D MDHistoWorkspace

zjmorgan, 2024-02-14:
**Describe the bug**

**To Reproduce**
1. Run this script
```python
from mantid.api import IMDHistoWorkspace

ws=CreateMDHistoWorkspace(Dimensionality=3,Extents='-3,3,-10,10,-4,4', \
                        SignalInput=range(0,1000),ErrorInput=range(0,1000),\
                        NumberOfBins='10,10,10',Names='Dim1,Dim2,Dim3',Units='MomentumTransfer,EnergyTransfer,Temperature')
```
2. Open slice viewer
3. Select ROI tool
4. Try to export cuts/slices with keyboard

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2024-02-19:
Possible duplicate of #36084?

sf1919, 2024-02-19:
If it is a duplicate is there a reason why it's suddenly high priority?

zjmorgan, 2024-02-19:
It is a duplicate

closed 2024-02-19 by zjmorgan

sf1919, 2024-02-19:
I would definitely add the `Reported by User` label to the other issue as a minimum.