mantidproject/mantid #36866: Windows Clean/Dirty Install Smoke Tests  :soap: :hankey:

jclarkeSTFC, 2024-02-20:
We must test both the .exe installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.8.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the new release candidate of Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Remove the mantid  files in %APPDATA%

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove (or rename) `~/Library/Preferences/org.python.python.Python.plist`  and `~/Library/Preferences/org.python.python.plist` 

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

Re-install the new release candidate of Mantid

- [ ] Check that Mantid boots up correctly

sf1919, 2024-02-21:
I have done both of these on Conda and Mantid booted correctly both times. Do we want someone to test this on a non-conda package too? @jclarkeSTFC @SilkeSchomann

SilkeSchomann, 2024-02-21:
> I have done both of these on Conda and Mantid booted correctly both times. Do we want someone to test this on a non-conda package too? @jclarkeSTFC @SilkeSchomann

Yes, that would be good.

MohamedAlmaki, 2024-02-21:
It works as expected on my machine

closed 2024-02-21 by MohamedAlmaki