mantidproject/mantid #36877: MacOS Python Smoke Tests  :snake:

jclarkeSTFC, 2024-02-20:
We must test both the .dmg installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.8.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :snake:

* Test that the Python scripting window works, [directions here](https://docs.mantidproject.org/nightly/workbench/scriptwindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/index.html), 3 or 4 examples from the Solutions is enough

perenon, 2024-02-21:
In the "Solution 2" example [here](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/solutions/02_pim_sol.html#b-plotting-ill-data), the `axes.legend().draggable()` code should be replaced by `axes.legend().set_draggable(True)`.

perenon, 2024-02-21:
@jclarkeSTFC, these issues are relatively small - only documentation related. We suggest to close this issue.

jclarkeSTFC, 2024-02-21:
Thanks Remi, yes it looks like we just need to update the documentation.

closed 2024-02-21 by jclarkeSTFC