mantidproject/mantid #36889: IDAaaS Interface Smoke Tests  :facepalm:

jclarkeSTFC, 2024-02-20:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] MantidWorkbench opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

jhaigh0, 2024-02-21:
Tiny bug. If you turn off plot titles in the settings, then open a plot, then change its name in the lower left plot widget, a title gets added. It should just update the window title and not add a plot title.

jhaigh0, 2024-02-22:
> Tiny bug. If you turn off plot titles in the settings, then open a plot, then change its name in the lower left plot widget, a title gets added. It should just update the window title and not add a plot title.

Issue here https://github.com/mantidproject/mantid/issues/36913

closed 2024-02-22 by SilkeSchomann