mantidproject/mantid #36905: plt.show() makes the editor not editable

perenon, 2024-02-21:
**Describe the bug**
If running `plt.show()` in a script editor, the editor comes to a state where it cannot be edited anymore. 

**To Reproduce**
Open mantidworkbench

In the editor, type:
```

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

# Load the data and extract the region of interest
data=Load('164198.nxs')
data=ExtractSpectra(data, XMin=470, XMax=490, StartWorkspaceIndex=199, EndWorkspaceIndex=209)

'''2D Plotting - Colorfill and Contour'''

# Get a figure and axes for
figC,axC = plt.subplots(ncols=2, subplot_kw={'projection':'mantid'}, figsize = (6,4))

# Plot the data as a 2D colorfill: IMPORTANT to set origin to lower
c=axC[0].imshow(data,cmap='jet', aspect='auto', origin = 'lower')

# Change the title
axC[0].set_title("Colorfill")

# Plot the data as a 2D colorfill: IMPORTANT to set origin to lower
c=axC[1].imshow(data,cmap='jet', aspect='auto', origin = 'lower')

# Overlay Contour lines
axC[1].contour(data, levels=np.linspace(0, 10000, 7), colors='white', alpha=0.5)

# Change the title
axC[1].set_title("Contour")

# Add a Colorbar with a label
cbar=figC.colorbar(c)
cbar.set_label('Counts ($\mu s$)$^{-1}$')

'''3D Plotting - Surface and Wireframe'''

# Get a different set of figure and axes with 3 subplots for 3D plotting
fig3d,ax3d = plt.subplots(ncols=2, subplot_kw={'projection':'mantid3d'}, figsize = (8,3))

# 3D plot the data, and choose colormaps and colors
ax3d[0].plot_surface(data, cmap='summer')
ax3d[1].plot_wireframe(data, color='darkmagenta')

# Add titles to the 3D plots
ax3d[0].set_title("Surface")
ax3d[1].set_title("Wireframe")

#plt.show()# uncomment to show the plots
```

Run the script, the plots appear. Uncomment the last line, run once again. The plot will also appear, but the editor will not be editable anymore. Closing the tab makes things back to normal.

**Expected behavior**

The editor should be editable.

**Screenshots**

**Platform/Version (please complete the following information):**
 - Tested on macOS - version 6.9.0rc1 installed with dmg installer
 - Tested on Ubuntu - version 6.9.0rc1 installed with conda

**Additional context**

perenon, 2024-02-21:
Works also with this script

```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

# Load the data
run = Load('Training_Exercise3a_SNS.nxs')

fig, axes = plt.subplots(subplot_kw={'projection': 'mantid'})

# Choose legend labels and colors for each curve
labels = ("sp-1", "sp-2", "sp-3", "sp-4", "sp-5")
colors = ('#FFCFC4', '#FE886F','#FE4A23', '#B82405', '#6A1300')

# Plot the first 5 spectra
for i in range(5):
    axes.plot(run, wkspIndex=i, color=colors[i], label=labels[i])

# Plot the 9th spectrum with errorbars
axes.errorbar(run, specNum=9, capsize=2.0, color='blue', label='Peak of Interest', linewidth=1.0)

# Set the X-axis limts and the Y-axis scale
axes.set_xlim(-1.5, 1.8)
plt.yscale('log')

# Give the plot a title
plt.title("Peak Evolution", fontsize=20)

# Add a legend with the chosen labels and show the plot
axes.legend()
#plt.show() #uncomment to show the plot

# Note with the Direct Matplotlib method,
# there are many more options for formatting the plot
```

AndreiSavici, 2024-03-01:
We should stop users from using `plt.show()` and ask them to use `fig.show()`

perenon, 2024-03-01:
Hi Andrei. 
Thanks, that solves the whole problem for the two issues described above.

Apparently, there is a PR [here](https://github.com/mantidproject/mantid/issues/36938) to change the code generation in that way. We should also update documentation.

jhaigh0, 2024-03-15:
This will be something to look into,

https://github.com/matplotlib/matplotlib/pull/23101

closed 2026-08-25 by github-actions[bot]