mantidproject/mantid #36938: Mantid Figure/plot script generator uses plt.show() not fig.show()

RichardWaiteSTFC, 2024-02-28:
**Describe the bug**
Mantid Figure/plot script generator uses `plt.show()` not `fig.show()`
It's been reported by users (incl. Pascal on WISH) that `plt.show()` causes mantid to hang.

**To Reproduce**
(1) `ws = CreateSampleWorkspace()`
(2) Plot a spectrum of `ws`
(3) Copy script to clipboard using button in toolbar

Produces 
```
fig, axes = plt.subplots(edgecolor='#ffffff', num='ws-3', subplot_kw={'projection': 'mantid'})
axes.plot(ws, color='#1f77b4', label='ws: spec 1', wkspIndex=0)
axes.tick_params(axis='x', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
axes.tick_params(axis='y', which='major', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
axes.set_title('ws')
axes.set_xlabel('Time-of-flight ($\\mu s$)')
axes.set_ylabel('Counts ($\\mu s$)$^{-1}$')
legend = axes.legend(fontsize=8.0).set_draggable(True).legend

plt.show()
```
not 
```
fig.show()
```

**Expected behavior**
Would have last line `fig.show()`

sf1919, 2024-03-01:
See #36310

thomashampson, 2024-03-19:
I've just tried this in 6.9.0 and it seems to be fine.

PR #36310 (2024-03-21): Fix for plot script unhiding other plots (switch plt.show() to fig.show())
### Description of work
The script generated by a plot window ends with a call to `plt.show()`, but `show()` causes all hidden plots to be unhidden. This was changed from calling `fig.show()` in #28699 because the `matplotlib` online documentation usually uses `plt.show()`. There's an explanation [here](https://stackoverflow.com/a/51570439) about the difference (assuming the random person who wrote that answer is correct). If someone is running the plot script outside of Workbench then `plt.show()` is more appropriate,  [... cut]
closed 2024-03-21 by robertapplin