mantidproject/mantid #36969: Inelastic Bayes Quasi crash with OSIRIS data

robertapplin, 2024-03-05:
**Describe the bug**
There is a crash on the Inelastic Bayes fitting Quasi tab when using OSIRIS data after clicking run.

**To Reproduce**
Load the following datasets

Sample
[osiris113015_multi_graphite002_red.zip](https://github.com/mantidproject/mantid/files/14494578/osiris113015_multi_graphite002_red.zip)

Resolution 
[osiris113015_graphite002_red.zip](https://github.com/mantidproject/mantid/files/14494577/osiris113015_graphite002_red.zip)

Click Run and wait. It will eventually crash

**Expected behavior**
It shouldn't crash. If necessary there should be a warning message.

**Platform/Version (please complete the following information):**
 - OS: Found on Windows

SpencerHowells, 2024-03-05:
@robertapplin Do not understand this data whatsoever! 1 - the 2 sets of data are the same sample - the multi version is usually a sum of runs for better statistics. 2 - the peak is at about 3.5 mev so is not QENS and Quasi should not be run. 3 - I've run Reduction & it appears to read a nxs raw file which is not the usual method. 4 - it only has 42 spectra wheres it is now 51. 5 - the run  date is 2015 so it may be a different instrument setup.   And I could go on!

robertapplin, 2024-03-05:
Agreed, the data doesn't make sense from a science perspective.

Ideally the software shouldn't crash no matter what we throw at it. I've done some investigating and the crash comes from a `STOP` statement in the `quasielasticbayes` fortran PyPi package. There is no way to "catch" these terminations, so the only way to prevent a crash like this would be to change the `quasielasticbayes` package to use exit status codes rather than `STOP`. This would be a lot of work and I would argue not worth the effort given that we hope to use the `quickBayes` package going forward
https://github.com/mantidproject/mantid/issues/35918

SpencerHowells, 2024-03-05:
@robertapplin Tried running BayesQuasi2 and the fitting crashed but not Mantid. Again due to bad data.

robertapplin, 2024-03-05:
Opening this PR https://github.com/mantidproject/mantid/pull/36973 so we can start testing quickBayes in the Bayes fitting interface

closed 2024-04-04 by robertapplin