mantidproject/mantid #37012: Integration range changes from Elwin GUI does not change the energy integration window when running the algorithm

adriazalvarez, 2024-03-13:
**Original reporter:** Franz Demmel

**Describe the bug**
As described by Franz
```
I just now noticed that when I use elwin the energy integration window does not change anymore.
Independent of the changed integration range in the GUI it seems always to integrate around the elastic line=the standard values.
```
**To Reproduce**

1. In Mantid Workbench, open `Interfaces->Inelastic->Data Manipulation` interface and select the `Elwin` tab.
2. On File, browse for file `irs26173_graphite002_red.nxs` and click on `Run`. 
3. Plot the resulting spectra by clicking on `Plot` button. 
4. Change the integration range either by the Property browser or directly using the graph delimiters on the plot widget. 
5. Run again the algorithm, see that the results are unaltered by the change of integration limits.

**Expected behavior**
GUI selectable integration limits should be taken into the model when running the `Elastic Window` algorithm.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

SpencerHowells, 2024-03-13:
@adriazalvarez It would be useful if Sample Logs in the results were populated with the energy range values.

adriazalvarez, 2024-03-13:
@SpencerHowells Noted, thank you.

adriazalvarez, 2024-03-14:
@SpencerHowells I will open a separate issue for adding the sample logs. It would easier because We will refactor some functionality of Elwin tab in the next sprint and will be a good time to implement this feature.

SpencerHowells, 2024-03-14:
@adriazalvarez OK

PR #37016 (2024-03-14): Fix sliders not changing integration limits in Elwin tab
### Description of work
This PR fixes #37012 , when elastic window integration limits are updated by using the slider on the plot, these limits were not passed to the presenter, and not updated in the model, so running the algorithm while changing the limits form the slider did not have any effect on the resulting computed elastic window. 
#### Summary of work

Fixes #37012 

#### Further detail of work

### To test:
1. In Mantid Workbench, open `Interfaces->Inelastic->Data Manipulation` interface and select the `Elwin` tab.
2. On File,  [... cut]
closed 2024-03-14 by thomashampson