mantidproject/mantid #37030: Plotting slow and ultimately crashing on Linux

sf1919, 2024-03-18:
**Original reporter:** [username facility]/[nobody]
Too many affected - see this support issue for those to contact once resolved.

**Describe the bug**
When trying to work with a lot of data or data that requires a lot of processing to plot then it seems to be very slow on Linux systems. So much so users are either getting a Force Quit/Terminate or Continue dialog box or the window freezes. If the window freezes then users trying to close the window will also get a Force Quit/Terminate or Continue dialog box. On choosing Force Quit/Terminate option Mantid crashes.

**To Reproduce**

1. Open Mantid on Linux
2. Load two files from the training dataset, MAR11015 and MAR11060
3. Right click one of the two files and select Plot->Superplot
4. Drag the other file onto the resulting Superplot and choose to plot all spectrum (should be 900+)
5. Move the plotting window whilst it's running
6. A `Not responding` dialog box comes up. Click terminate or force quit. Mantid crashes

If the dialog doesn't appear or disappears too quickly to terminate/force quit then try the following additional steps
7. Start trying to remove lines from the plot (click the `-` button by spectrums in the list on the plot)
8. The pane will probably freeze at this point so would won't be able to move it around etc. Click the `x` in the top right hand corner.
9. See step 6 above.

**Expected behavior**
There are a couple of things that would be good to resolve.

The first is that we don't necessarily want Mantid crashing everytime a user tries to force close a plotting pane.

The second is that it would be good for plotting in Mantid on Linux to not result in the `Not responding` dialog so frequently. More stable processing is required.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Ubuntu 20.04, CentOS Linux 7, Ubuntu 22.04.4, Rocky 8.9

 - Mantid Version: v6.9 and Nightly's after this point

**Additional context**
This is potentially a problem caused by upgrading matplotlib (#36351)

thomashampson, 2024-03-19:
The steps to reproduce cause exactly the same issue in version 6.8

jhaigh0, 2024-03-20:
We have tracked down how to reproduce what we think is the main cause of the plotting problems, I've written it up in this issue https://github.com/mantidproject/mantid/issues/37046

sf1919, 2024-03-28:
Given that we have managed to resolve the plotting issues we were facing and put out a patch release (v6.9.1) I think this one can be closed.

closed 2024-03-28 by sf1919