mantidproject/mantid #37039: Superplot not showing plot curves

GuiMacielPereira, 2024-03-19:
**Describe the bug**
Whilst following the instructions on #37030, I noticed that the superplot is not showing the curves. This is a regression from 6.8.

Current behaviour:
[image]

Expected behaviour:
[image]

**Platform/Version (please complete the following information):**
 - OS: Pop_OS (similar to Ubuntu)
 - OS Version: 22.04
 - Mantid Version [e.g. 6.0.0]

**Additional context**

thomashampson, 2024-03-19:
This is a duplicate of an issue I found last week:
https://github.com/mantidproject/mantid/issues/37007

sf1919, 2024-03-19:
As this is a duplicate I will close this one