mantidproject/mantid #37050: Throw error on invalid workspace names

peterfpeterson, 2024-03-20:
The [ADS exposes several methods](https://docs.mantidproject.org/nightly/api/python/mantid/api/AnalysisDataServiceImpl.html) where the names of workspaces become python variable names and is [described in the user documentation](https://docs.mantidproject.org/nightly/concepts/WorkingWithWorspaces.html). The general rules are ((taken from)[https://pythonguides.com/python-naming-conventions/])

* Names are case-sensitive
* Names cannot start with a digit
* Names can contain letters, numbers, and underscores

Note that unicode characters are allowed, but they may not be appropriate for workspace names because the ADS stores the names as `std::string`.

rboston628, 2025-07-17:
The sanitizer should also forbid `"/"` characters from names, for possible conflicts when saving workspace names as entry names in a Nexus file.  See Issue #39672

peterfpeterson, 2026-08-18:
Keep open