mantidproject/mantid #37132: Unhandled Exception When Monitor Workspaces Are Cropped

cailafinn, 2024-04-05:
**Describe the bug**
An unhandled exception occurs when a monitor workspace is cropped while the plot is open. For regular workspaces, cropping the workspace also updates the spectrum numbers associated with each curve, but for monitor workspaces they seem to keep spectrum 1 when it is removed. This causes an unhandled exception whenever the figure window is opened. 
**To Reproduce**

For example
1. Load something with monitors like LOQ49886 from the Training Course Data. Separate the monitors.
2. Plot the monitor workspace. 
3. Run CropWorkspace to remove the first spectrum (Put 1 in the StartWorkspaceIndex property).
4. Try to open the figure options on the plot. 

**Expected behavior**
The legend and lines should correctly represent the spectrum numbers present in the workspace, and allow the figure management widget to be opened. 

**Screenshots**
This screenshot shows the monitor workspace with it's missing first spectrum, but the legend shows a line with a spec num of 1. 
[image]

**Platform/Version (please complete the following information):**
 - OS: Found on macOS 14. Should be present on all OSs.

jhaigh0, 2024-06-06:
Stack trace I get when recreating this
```
Traceback (most recent call last):
  File "C:\Users\bya67386\work\mantid\qt\applications\workbench\workbench\plotting\figuremanager.py", line 365, in launch_plot_options
    self.plot_options_dialog = PlotConfigDialogPresenter(self.canvas.figure, parent=self.window)
  File "C:\Users\bya67386\work\mantid\qt\python\mantidqt\mantidqt\widgets\plotconfigdialog\presenter.py", line 45, in __init__
    curves_tab = CurvesTabWidgetPresenter(self.fig, parent_view=self.view, parent_presenter=self, legend_tab=legend_tab)
  File "C:\Users\bya67386\work\mantid\qt\python\mantidqt\mantidqt\widgets\plotconfigdialog\curvestabwidget\presenter.py", line 39, in __init__
    self.update_view()
  File "C:\Users\bya67386\work\mantid\qt\python\mantidqt\mantidqt\widgets\plotconfigdialog\curvestabwidget\presenter.py", line 287, in update_view
    self.set_errorbars_tab_enabled()
  File "C:\Users\bya67386\work\mantid\qt\python\mantidqt\mantidqt\widgets\plotconfigdialog\curvestabwidget\presenter.py", line 247, in set_errorbars_tab_enabled
    enable_errorbars = curve_has_errors(self.get_current_curve())
  File "C:\Users\bya67386\work\mantid\qt\python\mantidqt\mantidqt\widgets\plotconfigdialog\curvestabwidget\__init__.py", line 114, in curve_has_errors
    return ax.artists_workspace_has_errors(curve)
  File "C:\Users\bya67386\work\mantid\Framework\PythonInterface\mantid\plots\mantidaxes.py", line 324, in artists_workspace_has_errors
    workspace_index = workspace.getIndexFromSpectrumNumber(spec_num)
RuntimeError: Could not find spectrum number in any spectrum.
```

jclarkeSTFC, 2024-07-15:
Turns out that the only time this does work is if you type in a range of spectrum numbers to plot. If you type in workspace indices or click `Plot All` then you'll get the same exception, doesn't have to be a monitor workspace.

jclarkeSTFC, 2024-07-15:
When plotting spectra, if you either type in workspace indices, or press `Plot All`, then each line is plotted using its workspace index. If the corresponding workspace is then cropped then of course that index will be incorrect. Plotting by spectra numbers works because those do not change when a workspace is cropped. So an easy way to fix `Plot All` is to use this:
``` python
selection.spectra = self._plottable_spectra
```
instead of 
``` python
selection.wksp_indices = range(self.wi_min, self.wi_max + 1)
```
in `spectraselectordialog.py:on_plot_all_clicked()`. However, the original bug will still happen when specifying workspace indices. The code is set up to handle multiple workspaces, so if a workspace index always corresponds to the same spectrum in each of those workspaces, then what we could do is always plot with the spectra indices, i.e. convert any specified workspace indices to spectra indices before making the original plot.