mantidproject/mantid #37164: Use NexusHDF5Descriptor for HDF5 file loaders

robertapplin, 2024-04-12:


rboston628, 2025-11-14:
Some things moved around as part of the Nexus refactor (Roadmap Issue #38332).

`NexusHDF5Descriptor` was first renamed to `NexusDescriptor` in PR #38989 

Then `NexusDescriptor` was moved from `Kernel` to `Nexus` in PR #39278

The class formerly known as `NexusDescriptor` (which loads both HDF4/HDF5) was renamed to `LegacyNexusDescriptor` and is now *only* used for load algorithms related to muons files.  Work done in PR #38824.

All other load algorithms use `NexusHDF5Descriptor` as of PR #38786 (which was later renamed to `NexusDescriptor`).

**However** that required refactoring `NexusHDF5Descriptor` to load *all* entries, which undoes the purpose of this issue.

So in one sense all the work was done, as now everything uses `NexusHDF5Descriptor`.  However, in another sense all the work was undone because `NexusHDF5Descriptor` loads the entire file tree, which was deemed a wasteful operation at the file tree level.

I would propose something like a `NexusTopLevelDescriptor` or `NexusShallowDescriptor` for use in the `Load` and related algos, intended for these confidence calculations, which only loads the top one or two levels (need an audit how deep in the tree anything goes).

peterfpeterson, 2026-01-13:
At the tip of the branch for #40614 I ran the attached script [fileregistrybenchmark.py](https://github.com/user-attachments/files/24600870/fileregistrybenchmark.py) (put it in your build directory, maybe adjust the directory to search) which finds the time spent scoring every file in the system tests
```
0.4010  / 125 = 0.0032 LoadNexusProcessed
0.0138  /  58 = 0.0002 LoadRaw
1.3591  /  49 = 0.0277 LoadEventNexus
0.0007  /   2 = 0.0004 LoadGSS
0.0100  /  26 = 0.0004 LoadAscii
0.8630  /  20 = 0.0432 LoadISISNexus
0.0011  /   3 = 0.0004 LoadDaveGrp
0.0429  /   2 = 0.0215 LoadMuonNexusV2
0.0021  /   5 = 0.0004 LoadFITS
0.0004  /   1 = 0.0004 LoadSpec
0.0142  /   5 = 0.0028 LoadMD
0.0003  /   1 = 0.0003 LoadNXSPE
0.0003  /   1 = 0.0003 LoadSINQFocus
0.0304  /  14 = 0.0022 LoadMuonNexus
0.0007  /   2 = 0.0004 LoadIsawPeaks
0.0014  /   1 = 0.0014 LoadHFIRSANS
0.0003  /   1 = 0.0003 LoadSPE
0.0004  /   1 = 0.0004 LoadEventPreNexus
```
formated from

```
<total time when this loader wins> / <times it won> = <time per win> <loader name>
```

The worst offenders on time per win are LoadISISNexus, LoadEventNexus, LoadMuonNexusV2. Everything else is 10x faster than those 3.

rboston628, 2026-01-16:
Digging into this more.  At around the time this issue was created, these links show the state of the two descriptors:
- [`NexusDescriptor` ca April 2024](https://github.com/mantidproject/mantid/blob/8e8b9d49956818f8e2e300f98d18df8c4ef7c104/Framework/Kernel/src/NexusDescriptor.cpp)
- [`NexusHDF5Descriptor` ca April 2024](https://github.com/mantidproject/mantid/blob/8e8b9d49956818f8e2e300f98d18df8c4ef7c104/Framework/Kernel/src/NexusHDF5Descriptor.cpp)

Importantly, `NexusDescriptor` at that time was using `::NeXus::File::getEntries()`, as it is distributed in `napi`.  That will ultimately call [`NX5getNextEntry`](https://github.com/nexusformat/code/blob/5b803b3a0014bd9759b3d846da3cd3c1cfafd7d5/src/napi5.c#L1702).  Multiple places in the Mantid codebase note how slow it is.  However, `NexusHDF5Descriptor` works with the hdf5 C-API directly, including using the same low-level iteration procedures now in place for filling in the cache map.

Since the Nexus refactor work, Mantid's implementation of `Nexus::File::getEntries()` was greatly optimized to run on timescales similar to above.  Also, `NexusDescriptor` was renamed to `LegacyNexusDescriptor` and the `NexusHDF5Descriptor`.

The main work for this seems to have been accomplished, not by reducing the depth of the search, but by not using the old `getEntries()` method.  As of 6.14, the time it takes to load that file from the `Load` algorithm (assuming we correctly identified) is down to under 1s, total, on my laptop.

That said, we have reduced times even further by implementing the `NexusDescriptorLazy` object for file confidence scoring.