mantidproject/mantid #37213: Cannot load ISIS event data file

mducle, 2024-04-24:
**Original reporter:** ISIS / M. Aouane

**Describe the bug**

Mantid gives an error: 

```
Error in execution of algorithm LoadEventNexus:
Cannot guess ending value from a TimeSeriesProperty that contains only a single time
```

when trying to load this particular file:

```
from mantid.simpleapi import *
Load(Filename='/data/instrument/MARI/CYCLE20241/RB0/MAR29527.n001', OutputWorkspace='MAR29527')
```

The file should be publicly accessible on the DAaaS system - or I can send directly (it's a bit big so I don't want to attach it to this issue).

**To Reproduce**

Run the script above

**Expected behavior**

The file loads without error.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Rocky 8 Linux
 - OS Version:
 - Mantid Version: 6.9.1 (also checked on 6.7 and 6.8 which give the same error).

**Additional context**

The file loads fine with [nexpy](https://github.com/nexpy/nexpy) and [hdfview](https://www.hdfgroup.org/downloads/hdfview/) so is a valid NeXus file. The error seems to be in the [RoI (region of interest) processing](https://github.com/mantidproject/mantid/blob/5ecdbb541567066bbaceafc83d5e7450293f990c/Framework/DataHandling/src/LoadEventNexus.cpp#L1233-L1235) in `LoadEventNexus` which in turn calls [TimeROI::replaceROI](https://github.com/mantidproject/mantid/blob/5ecdbb541567066bbaceafc83d5e7450293f990c/Framework/Kernel/src/TimeROI.cpp#L396) which gives the error, but I can't figure out the bug from just glancing at the code - all of the log properties in the file have more than one entry... so it seems something wrong in the logic in `LoadNexusEvent`.

mducle, 2024-04-30:
Fixed in nightly `6.9.1.20240425.1854`.

GuiMacielPereira, 2024-05-01:
Reopening just to add that I suspect it was the work on https://github.com/mantidproject/mantid/pull/37141 that unintentionally fixed it

closed 2024-05-01 by GuiMacielPereira