mantidproject/mantid #37234: Expose WorkspaceNearestNeighbourInfo to python

RichardWaiteSTFC, 2024-04-29:
**Describe the outcome that is desired.**
Exposing `WorkspaceNearestNeighbourInfo` 
https://github.com/mantidproject/mantid/blob/0bd2ff9fc75ee7d52886bc22af0de638ebd18c6c/Framework/API/inc/MantidAPI/WorkspaceNearestNeighbourInfo.h#L25
to python would help in developing clustering algorithms for peak finding and perhaps also speed up detector/TOF integration methods such as `IntegratePeaksSkew` - which currently has some custom code for `ComponentArray` like detectors that is not very general (works for WISH only)

https://github.com/mantidproject/mantid/blob/0bd2ff9fc75ee7d52886bc22af0de638ebd18c6c/Framework/PythonInterface/plugins/algorithms/IntegratePeaksSkew.py#L114

RichardWaiteSTFC, 2024-11-25:
This would involve replacing the code in `_get_detid_array_comp_assembly` for WISH-type instrument IDF (i.e. `ComponentArray`), there is similar code for `RectangularDetector` instruments that currently works, but `WorkspaceNearestNeighbouInfo` may or may not be quicker - use whatever is quickest!

warunawickramasingha, 2025-06-19:
> This would involve replacing the code in `_get_detid_array_comp_assembly` for WISH-type instrument IDF (i.e. `ComponentArray`), there is similar code for `RectangularDetector` instruments that currently works, but `WorkspaceNearestNeighbouInfo` may or may not be quicker - use whatever is quickest!

Note: As agreed in the stand-up, the [pull request](https://github.com/mantidproject/mantid/pull/39499) only covers exposing WorkspaceNearestNeighbourInfo object to python and its unit tests.

closed 2025-06-24 by SilkeSchomann