mantidproject/mantid #37236: Load Algorithm no longer sets LoaderName

walshmm, 2024-04-29:
**Describe the bug**

When attempting to use the Load algorithm, if you do not set the LoaderName it used to set it for you once it ascertained the correct loader.  Now it does not.
[SNAPRed](https://github.com/neutrons/SNAPRed/) currently depends on this feature.

Based on my initial investigation, specifically [these lines](https://github.com/mantidproject/mantid/commit/117047e2c648750ef8e5d6f5b307b6ccb6012de8#diff-c2b373b035fb1f1debadccc5ca62f9a0b6750735af530723f5fba6b35797a425R346-R350) cause issue because now it doesn't call getFileLoader which is what sets the LoaderName [here](https://github.com/mantidproject/mantid/blob/117047e2c648750ef8e5d6f5b307b6ccb6012de8/Framework/DataHandling/src/Load.cpp#L199)

**To Reproduce**

[image]

**Platform/Version (please complete the following information):**
 - OS: all?
 - OS Version:
 - Mantid Version mantid                    6.9.20240425.1854 py310hf6feea2_0    mantid/label/nightly

**Additional context**

robertapplin, 2024-04-29:
The `LoaderName` should get set when the `Filename` property of `Load` gets set here
https://github.com/mantidproject/mantid/blob/117047e2c648750ef8e5d6f5b307b6ccb6012de8/Framework/DataHandling/src/Load.cpp#L118

Do you know if this part of the code gets entered in your test? Or maybe the `LoaderName` is somehow getting unset?

robertapplin, 2024-04-29:
Might it be that this line is overriding the found `LoaderName` with a blank string?
https://github.com/neutrons/SNAPRed/blob/4f3c0d1353d49a9fb21bc5f38d561454207a26e7/src/snapred/backend/recipe/algorithm/FetchGroceriesAlgorithm.py#L123

walshmm, 2024-04-29:
```
INFO     Mantid:FetchGroceriesRecipe.py:58 Parameters:
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: Filename, Value: /home/runner/work/SNAPRed/SNAPRed/tests/resources/inputs/test_555_fetchgroceriesrx.nxs, Default?: No, Direction: Input
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: OutputWorkspace, Value: _fetched_grocery, Default?: No, Direction: Output
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: LoaderType, Value: , Default?: Yes, Direction: InOut
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: LoaderArgs, Value: , Default?: Yes, Direction: Input
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: InstrumentName, Value: , Default?: Yes, Direction: Input
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: InstrumentFilename, Value: , Default?: Yes, Direction: Input
INFO     Mantid:FetchGroceriesRecipe.py:58   Name: InstrumentDonor, Value: , Default?: Yes, Direction: Input
```
Most of the inputs are defaults

walshmm, 2024-04-29:
This was discovered on a pr that did not touch any of this code, it worked fine/works fine on older pulls of mantid

walshmm, 2024-04-29:
[mantid37236_load_loadername_issue_logs.txt](https://github.com/mantidproject/mantid/files/15151979/mantid37236_load_loadername_issue_logs.txt)
A dump of the logs

robertapplin, 2024-04-29:
Yes, so this change was made because Mantid was being a bit inefficient by searching for which Load algorithm to use twice. It was searching once when the `Filename` property was set, and then a second time when `execute` was called. In some cases this caused the loading time to be 33% slower than it needed to be #37150

The solution we chose was to cache the load algorithm which was found when `Filename` property was set, and then not to do the search a second time. It looks like fixing this has led to your PR failing because:

1. It sets the Filename property, and finds the `LoaderName` on line 121
https://github.com/neutrons/SNAPRed/blob/4f3c0d1353d49a9fb21bc5f38d561454207a26e7/src/snapred/backend/recipe/algorithm/FetchGroceriesAlgorithm.py#L121
2. It then sets the LoaderName property to `loaderType` on line 123 (which is an empty string), so it will override the loader that has just been found
https://github.com/neutrons/SNAPRed/blob/4f3c0d1353d49a9fb21bc5f38d561454207a26e7/src/snapred/backend/recipe/algorithm/FetchGroceriesAlgorithm.py#L123
3. When execute is called, it no longer does a second search for a loader because it has already got a cached loader. So the loadertype now comes back as blank

walshmm, 2024-04-29:
So could we just set the loaderType property regardless of cache hit?

walshmm, 2024-04-29:
Btw we were going to implement a fix for this ourselves, please let us know if you had planned on picking this up so we can coordinate.

rboston628, 2024-04-29:
We suspect it was this commit: https://github.com/mantidproject/mantid/commit/117047e2c648750ef8e5d6f5b307b6ccb6012de8
I think the issue is just that `loaderName` isn't being set here.

robertapplin, 2024-04-29:
> Btw we were going to implement a fix for this ourselves, please let us know if you had planned on picking this up so we can coordinate.

I'm happy for you to pick this up. This is a breaking change that will be part of v6.10 when it gets released, however I think its working as it would be expected now. Personally I would suggest that the `LoaderName` does not get set on https://github.com/neutrons/SNAPRed/blob/4f3c0d1353d49a9fb21bc5f38d561454207a26e7/src/snapred/backend/recipe/algorithm/FetchGroceriesAlgorithm.py#L123 
if it is blank

rboston628, 2024-04-29:
I just tried Rob's suggestion, it seemed to have worked

walshmm, 2024-04-29:
The above mentioned change gets us unstuck, if no one else depends on this behavior I'm fine to let sleeping dogs lie.

walshmm, 2024-04-29:
Otherwise I feel this would be a minimal one line fix to just always return what loader it used.

rboston628, 2024-04-29:
I tried running this script in my dev build of mantid from `main`:

``` python
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

workspace = mtd.unique_name(prefix="loadTest_")
filename = os.path.expanduser("~/test_file.nxs.h5")

# create an easily readable event file
CreateWorkspace(
    OutputWorkspace=workspace,
    DataX=[0.5, 1.5],
    DataY=[0],
    DataE=[0.01],
    NSpec=1,
)
SaveNexus(
    Filename=filename,
    InputWorkspace=workspace,
)

# read in that file -- should use the LoadNexusProcessed loader
res = Load(
    Filename=filename,
    OutputWorkspace=workspace,
)
print(res)
```

This will create a small test file on the users home directory then open it.  According to the [algorithm information](https://docs.mantidproject.org/nightly/algorithms/Load-v1.html), there should be three outputs from this algorithm.  However, this script is only showing a single output, the workspace name.

More strange, our program SNAPRed, building off of `nightly`, is receiving three outputs from `Load`, the second one being the `LoaderType`.

(And also `Load` doesn't recognize `.nxs` as a valid file extension)

PR #37238 (2024-05-01): Reset loader name and version at end of algorithm
### Description of work

#### Summary of work

The `Load` algorithm searches for the best loading algorithm to use to load a file, then returns the loaded workspace, the name of the loader used, and the version of the loader used.

A minor issue allowed a user to overwrite the name and version of the loader, resulting in invalid returns on those outputs.

This makes sure the loader name and version are correctly set at the end of the algorithm.

Fixes #37236 . 

#### Further detail of work
The `Load` algorithm terminates with a call to set the  [... cut]
closed 2024-05-01 by SilkeSchomann