mantidproject/mantid #37252: Segfault when Rebinnnig in Instrument View

GuiMacielPereira, 2024-05-01:
**Original reporter:** Duc  duc.le@stfc.ac.uk

**Describe the bug**

Mantid crashes with segmentation fault when rebinning in Instrument View

**To Reproduce**
- Open Instrument View
- Go to Rebin
- Mistype it first with two numbers (e.g. `6500,18000`)
- An error will appear, close it
- Now type correctly with 3 numbers (e.g. `6500, 100, 18000`
- Mantid will crash with segfault

I tried it with MAR11001 and was able to reproduce

**Platform/Version (please complete the following information):**
 - OS Version: Rocky Linux 8.9 (Green Obsidian)
 - Mantid Version 6.9.1

**Additional context**

Report back to Duc when fixed

robertapplin, 2024-05-01:
@SilkeSchomann / @thomashampson I'm not sure if this should be an issue for Excitations or for Core?

GuiMacielPereira, 2024-05-01:
@thomashampson  and I just tried to reproduce it but couldn't make it to crash... So maybe we can just close it?

jclarkeSTFC, 2024-05-14:
I just tried to reproduce this on IDAaaS (no GPU), but if OpenGL is enabled then Workbench vanishes (#36220) when I try and open the instrument view. If I disable OpenGL, then I can open the instrument view, but I can't reproduce the crash in this issue. I'm not saying necessarily that this problem is to do with OpenGL, just that I couldn't test it with OpenGL enabled.

jclarkeSTFC, 2024-05-14:
I also just tried this on an IDAaaS workspace with a GPU (so OpenGL enabled, #36220 only applies when there's no GPU), and I couldn't reproduce this Rebin crash.

cailafinn, 2024-05-14:
I've tried this on macOS. With a variety of files and with openGL enabled and disabled I still could not reproduce the crash. Mantid opened with this in the error log, so it's possible openGL things are handled differently on mac:
```
An OpenGL surfcace format was requested that is either not version 3.2 or higher or a not Core Profile.
Chromium on macOS will fall back to software rendering in this case.
```