mantidproject/mantid #37253: Missing images in online docs

adriazalvarez, 2024-05-01:
Indirect Difraction section of the documentation contains some missing images. 

**Expected behavior**

Images are loaded correctly on the docs.
**Screenshots**

[image]
**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

sf1919, 2024-05-01:
Can you check whether these are showing on the online documents? The developer setup of CMake does not automatically download images which is why it mentions enabling screenshots.

adriazalvarez, 2024-05-01:
No, it's not a local build. I think it's on the server build. I have added the link to the page in the issue.

PR #37266 (2024-05-03): Fix missing screenshots for Indirect Diffraction
### Description of work
This PR fixes some missing screenshots on the Indirect Diffraction documentation page: https://docs.mantidproject.org/v6.1.0/interfaces/indirect/Indirect%20Diffraction.html

I removed one of the screenshots because it wasn't helpful, and I fixed the other screenshot:
[image]

Fixes #37253 

### To test:

1. Go to your build folder, open `CMakeCache.txt` and set `DOCS_SCREENSHOTS` to `ON`.
2. From git bash, cd to build folder, and run `cmake .` to re-configure
3. Open visual studio
4. Build `ALL_BUILD`
5. Build `docs-html` [... cut]
closed 2024-05-03 by SilkeSchomann