mantidproject/mantid #37294: Improve test code for PolarizationCorrectionWildes

rbauststfc, 2024-05-08:
The unit test code for `PolarizationCorrectionWildes` would benefit from reducing some of the code duplication and potentially improving some of the readability to help us maintain and extend it over the longer term. We expect this algorithm to be used more and more by SANS and Reflectometry scientists at ISIS, so this may help us fix and make changes more easily in the future.

We could also consider adding a system test as part of this work.

jclarkeSTFC, 2024-05-24:
For the system test, do we have any particular examples to use?

rbauststfc, 2024-05-24:
I'm afraid not, identifying what we can use was part of what needs doing under this issue. There is some POLREF test data already available, but I'm not sure if we have both PNR (2 periods/spin states) and PA (4 periods/spin states) data. We would need to ask the scientists for any test data that we don't already have, and confirm with them that it's OK for the data to be made public. Depending on how you prefer to do things, if it works better, the system test part of this issue could be pulled out into a separate issue.

closed 2024-06-04 by SilkeSchomann