mantidproject/mantid #37298: Investigate why Doxygen Warnings are not failures

sf1919, 2024-05-08:
**Describe the outcome that is desired.**
At the moment the Doxygen pipeline will pass if there are only warnings (e.g. https://builds.mantidproject.org/view/Pull%20Requests/job/pull_requests-doxygen/52117/). Instead we want the build to register as a failure to prevent errors getting through.

**Describe any solutions you are considering**
Initially we thought we could change this by changing the Thresholds in the Jenkins interface (see image below) but this is not part of the configuration of the Doxygen pipeline on Jenkins. Another option will need to be found. 

[image]

**Additional context**

peterfpeterson, 2024-05-08:
I don't know if it makes a difference, but [WARN_LOGFILE](https://www.doxygen.nl/manual/config.html#cfg_warn_logfile) is not set in `Framework/Doxygen/Mantid_template.doxyfile`. I don't know if it matters, but the format of the printed warnings is set to
```
  # Put the warning format in something sensible for the generator we are using
  if(MSVC)
    # Allows Visual Studio to mark them as if they were compiler warnings and jump straight to them
    set(DOXY_WARN_FORMAT
        "\"$file($line): $text\""
        CACHE STRING "Warning format for doxygen output, which must include surrounding double quotes"
    )
  else()
    set(DOXY_WARN_FORMAT
        "\"$file:$line: $text \""
        CACHE STRING "Warning format for doxygen output, which must include surrounding double quotes"
    )
  endif()
```

closed 2024-05-08 by robertapplin

sf1919, 2024-05-09:
Am changing the milestone as it got in before code freeze.

PR #37302 (2024-05-08): Treat Doxygen warnings as errors
### Description of work

#### Summary of work
Doxygen warnings are now treated as errors after setting the corresponding flag in Framework/Doxygen/Mantid_template.doxyfile.

Fixes #37298.

### To test:

As there are no genuine Doxygen warnings at the moment, I created a temporary commit to trigger a Doxygen warning: [Added temporary modification to trigger warning](https://github.com/mantidproject/mantid/pull/37302/commits/4fdcf70953ba35c40c407110a878e99eb78324c4)

Please check that the Doxygen pipeline failed for this commit, but was successful after removing  [... cut]