mantidproject/mantid #37322: Manual Testing Indirect interfaces

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at: 

* https://developer.mantidproject.org/Testing/Indirect/DataReductionTests.html

adriazalvarez, 2024-05-15:
Documentation needs update. Some output workspace names are not correct.

RichardWaiteSTFC, 2024-05-15:
Test 1 Step 7 - missing instruction to check scale by factor
Test 1 Step 19 - need another step to click run
Test 2 Step 6 - mantid crashed when finding files

adriazalvarez, 2024-05-16:
I've opened an issue for updating documentation. 
I couldn't reproduce the issue with crashing.

closed 2024-05-16 by adriazalvarez