mantidproject/mantid #37323: Manual Testing Inelastic interfaces

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at:

* https://developer.mantidproject.org/Testing/Inelastic/QENSFittingTests.html 
* https://developer.mantidproject.org/Testing/Inelastic/DataManipulationTests.html

jhaigh0, 2024-05-15:
In Elwin Tab test 1
"In the main GUI right-click on `MAR27691-27698_graphite002_red_elwin_eq2` and choose Plot Spectrum, choose Plot All"

I think the file should be `MAR27691-27698_red_elwin_eq2`

jhaigh0, 2024-05-15:
Elwin tab test 2, had a crash when I clicked run.
At first, I added the new workspace with the previous 2 still loaded and clicked run but got an error. So I closed the interface, cleared the ads and followed the instructions from there (resulting in the crash).

jhaigh0, 2024-05-15:
In Indirect test 2, 
"Load the resolution file `irs26173_graphite002_res.nxs` from the sample data"
This file isn't in the sample data I think it should be `irs26174_graphite002_res.nxs`

jhaigh0, 2024-05-15:
For the MSD Fit tab test, when trying to add the ``irs26173_graphite002_res.nxs`` file I get the message "fitting data must contain one or more spectra"

jclarkeSTFC, 2024-05-16:
Data manipulation 15 mins

jclarkeSTFC, 2024-05-16:
At one point in the MSD test the interface got itself into a state where everything gave an error, but after changing a few options like function and minimiser it eventually sorted itself out. I couldn't reproduce the error.

closed 2024-05-20 by jclarkeSTFC