mantidproject/mantid #37325: Manual Testing Workbench Group 1- 4

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

RichardWaiteSTFC, 2024-05-16:
Issue with sliceviewer not rebinning if slicepoint moved within a bin (even a MDEvent workspace is binned by default with 100 bins along each axis) - it should trigger a rebin if an MDEvent workspace is present. Difficult to show clearly - will work on test instructions.

RichardWaiteSTFC, 2024-05-16:
> Issue with sliceviewer not rebinning if slicepoint moved within a bin (even a MDEvent workspace is binned by default with 100 bins along each axis) - it should trigger a rebin if an MDEvent workspace is present. Difficult to show clearly - will work on test instructions.

Issue raised for this #37325 - it is not a regression (seen also in v6.9) so have set milestone to 6.11

adriazalvarez, 2024-05-17:
We were testing for approximately 2 hours.

closed 2024-05-17 by adriazalvarez