mantidproject/mantid #37329: Manual Testing Engineering Diffraction

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

GuiMacielPereira, 2024-05-16:
When following Test 1 in the focus part, two separate plots were produced instead of a single plot with the two curves

GuiMacielPereira, 2024-05-16:
It would be good to introduce a comment at the top of the manual testing instructions warning about the difficulty with loading ENGINX files from the archive

GuiMacielPereira, 2024-05-16:
I also noticed that in Mantid the instrument is called ENGIN-X, but in the data cache its called ENGINX, will need to modify the data cache to account for this

GuiMacielPereira, 2024-05-16:
When undocking the plot from the Fitting tab, the plot settings bar stays docked

GuiMacielPereira, 2024-05-16:
On test 5, it's not clear that what is being tested is the browse option in the interface

GuiMacielPereira, 2024-05-16:
When adding a peak in the Fit browser, if you add a non BackToBackExponential peak and then change it to a BackToBackExponential peak, the default ties are not applied

GuiMacielPereira, 2024-05-16:
When fitting two BackToBackExponential peaks, we only saw the output workspaces for one peak, is this expected? When two peaks have the same fitting function, should the workspaces in the workspace group with the suffix `_fits` have the output parameters for both peaks?

GuiMacielPereira, 2024-05-16:
In the last step in Test 8, on `Fit > Evaluate Function`, we got a failure to converge. Maybe the testing instructions could be clearer on if this is expected?

GuiMacielPereira, 2024-05-16:
In Test 9, it's not clear where the primary log is and where it can be changed

GuiMacielPereira, 2024-05-16:
When we got a fit error, there is an error box with no text on it

RichardWaiteSTFC, 2024-05-16:
> When following Test 1 in the focus part, two separate plots were produced instead of a single plot with the two curves. We were not able to reproduce this later, though, so assume it was user error!

It can open multiple plots if you've selected the texture grouping? Or clicked the button twice!

RichardWaiteSTFC, 2024-05-16:
> When undocking the plot from the Fitting tab, the plot settings bar stays docked. It's unclear if this a known/desired behaviour however it looks like it's been the case since as far back as Mantid 6.7 (if not earlier).

This has always been the case, but I agree it would be nice it the toolbar stayed with the plot! If the scientists agree (hopefully I'll see them next week) I will make an issue for it.

RichardWaiteSTFC, 2024-05-16:
> When adding a peak in the Fit browser (test 8, step 5), if you add a non BackToBackExponential peak and then change it's type in the Fit Functions property browser to BackToBackExponential peak, the default ties are not applied. This is not a regression (appears in Mantid 6.7, perhaps earlier).

Good spot, I didn't know you could do this! This is not specific to the engineering interface, it is the same in the normal fit browser. It will be because it does not call `setMatrixWorkspace` on the function. Given the engineering UI sets the default peak shape to `BackToBackExponential` I don't think many users will come across this, but if you want to make a Core issue for it feel free - depending on how the release goes I may have time to pick it up.

RichardWaiteSTFC, 2024-05-16:
> When fitting two BackToBackExponential peaks on the same plot, we only saw the output workspaces for one peak, is this expected (perhaps it's not a realistic thing to try)? When two peaks have the same fitting function, should the workspaces in the workspace group with the suffix `_fits` have the output parameters for both peaks? The behaviour appears to be the same in Mantid 6.7.

There will be a single workspace for each peak parameter but it will have 2 columns (one for each peak) - like this
[image]

RichardWaiteSTFC, 2024-05-16:
> In the last step in Test 8, on `Fit > Evaluate Function`, we got a failure to converge. I don't seem to get the same thing if I go through the same steps in Mantid 6.9.1. I also notice in 6.9.1 that when you add a BackToBackExponential peak the A and B parameters seem to be automatically tied, whereas now they are automatically fixed. Is this all the expected behaviour now? If the failure to converge error is expected now then it might be good to state this in the testing instructions.

The failure to converge may be because to evaluate a function it runs runs `Fit` with 0 iterations - so it will not be converged! Again this is the case in the normal mantid fitbrowser, so not specific to the engineering UI. There is an algorithm called `EvaluateFunction` which probably wouldn't throw this warning, you could open a Core issue for this if you have the energy!

Also the parameters `A` and `B` should be fixed

RichardWaiteSTFC, 2024-05-16:
> fit error, we noticed what we thought initially was an error box with no text in it underneath the fit property window. After that we ran a fit that was successful and it showed a filled green strip for successful completion, instead of just the red border that was displayed for the error. At this point we realised that this was a progress bar, so we assume there is nothing that needs to be done here.

Yes just a progress bar, but a very thin one...

RichardWaiteSTFC, 2024-05-16:
@rbauststfc - thanks for testing this everyone, looks like some good suggestions for the testing docs. I don't think there are any other issues that need raising for the engineering UI, though some Core issues could be raised for generic fit browser bugs/unexpected behaviour.

closed 2024-05-20 by rbauststfc