mantidproject/mantid #37331: Manual Testing ISIS Reflectometry

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

koshchii, 2024-05-16:
In Section Loading Batch Files the instruction for Expand All Rows should be updated to Expand All Groups

koshchii, 2024-05-16:
We weren't able to find a good example to check the following task in "Copy, Cut and Paste": Select 2 rows from one group, copy them, and paste onto 2 rows in another group.

koshchii, 2024-05-16:
Need instructions update in "Save tab" section bullet number 6: Check that option Additional columns (includes Q resolution) is **unticked** but disabled, as it is not applicable.

SilkeSchomann, 2024-05-16:
In https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#id16 we could not find `ReflectometryISISSumBanks` in the workspace history.

koshchii, 2024-05-16:
it took us 1h 45 mins

koshchii, 2024-05-17:
I've opened an issue for updating documentation

rbauststfc, 2024-05-20:
> In https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#id16 we could not find `ReflectometryISISSumBanks` in the workspace history.

Thanks all for the testing! On this one, it all seems to be working OK for me. You have to have a region selected on the instrument view plot for `ReflectometryISISSumBanks` to be called, in case that helps explain what may have happened. Is this something you want to double check (I'm happy to run through it on a call with someone if that would help), or are we happy to conclude that it's OK and close this issue?

SilkeSchomann, 2024-05-20:
> > In https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#id16 we could not find `ReflectometryISISSumBanks` in the workspace history.
> 
> Thanks all for the testing! On this one, it all seems to be working OK for me. You have to have a region selected on the instrument view plot for `ReflectometryISISSumBanks` to be called, in case that helps explain what may have happened (bullet point 7 in the Preview Tab instructions is trying to make sure this is done). Is this something you want to double check (I'm happy to run through it on a call with someone if that would help), or are we happy to conclude that it's OK and close this issue?

I would be happy to close the issue. We thought it is probably caused by missing out on something in the test and not a genuine bug.

SilkeSchomann, 2024-05-20:
There is a separate issue for updating the documentation, so I am closing the test issue.

closed 2024-05-20 by SilkeSchomann