mantidproject/mantid #37332: Manual Testing ISIS SANS

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

cailafinn, 2024-05-15:
Crash when Following User Files instructions (5):
```shell
SANS-[Notice] Instrument changed to LOQ.
SANSLoad-[Notice] SANSLoad started
FrameworkManager-[Notice] Welcome to Mantid 6.9.20240514.1920
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
DownloadInstrument-[Notice] All instrument definitions up to date
CheckMantidVersion-[Notice] CheckMantidVersion started
CheckMantidVersion-[Notice] CheckMantidVersion successful, Duration 0.05 seconds
qt.qpa.fonts: Populating font family aliases took 123 ms. Replace uses of missing font family ".SF NS Text" with one that exists to avoid this cost. 
^CTraceback (most recent call last):
  File "/Users/caila.finn/mambaforge/envs/mantid_nightly/lib/python3.10/site-packages/mantidqt/dialogs/errorreports/report.py", line 144, in set_plain_text_edit_field
    def set_plain_text_edit_field(self):
KeyboardInterrupt
/Users/caila.finn/mambaforge/envs/mantid_nightly/bin/pythonw: line 3: 21670 Abort trap: 6           /Users/caila.finn/mambaforge/envs/mantid_nightly/python.app/Contents/MacOS/python "$@"
```

RichardWaiteSTFC, 2024-05-15:
Processing Test Step 7 - did not save can for second run

cailafinn, 2024-05-15:
I couldn't reproduce the crash again on IDAaaS or macOS. Weird. Will have to wait to see if it comes up again I think. 

Saving of the can turns out to be intentional behaviour. As the can outputs are the same, it saves time by only doing it once. If you change one of the lines to be a different can calculation then it outputs for both. I've made an issue for updating the manual testing instructions with this information:
- https://github.com/mantidproject/mantid/issues/37373

closed 2024-05-15 by cailafinn