mantidproject/mantid #37335: Manual Testing M-slice

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Direct/MSliceTestGuide.html

jclarkeSTFC, 2024-05-15:
On macOS Big Sur (11.5.2), in "Plotting a Slice" step 2, in the "Taking Slices" section, when the plot appears it is stuck behind the mSlice window, and I can't get the figure window to appear on top.

jhaigh0, 2024-05-15:
On IDAaaS, when clicking on the MSlice window, plot windows get hidden / pushed to the back (I have workbench open fullscreen behind it)

jhaigh0, 2024-05-15:
In test 6 - Generating a plot script. I had two plots open, one was marked keep and was a full view of the data, one was marked Make Current and was cropped in. I changed lots of plot options on the first 'Keep' plot, generated a script. When I ran the script it seemed to apply all the plot options to the 'make current' plot, as in it was cropped in but had all the changes I made to the 'keep' plot (log scale changes, grid turned on etc.)

jhaigh0, 2024-05-15:
5. Overplot Bragg Peaks, not getting the aluminium lines on the line plot.

jhaigh0, 2024-05-15:
If you have a workspace X, and use compose->scale to scale it and create X_scaled. If you scale X again but with a different value, I think it silently replaces X_scaled instead of making X_scaled_2

jclarkeSTFC, 2024-05-15:
Took about 45-60 mins

closed 2024-05-20 by jclarkeSTFC