mantidproject/mantid #37336: Manual Testing Project Recovery

koshchii, 2024-05-13:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

GuiMacielPereira, 2024-05-15:
We ran test 5 but kept the instance that hadn't crashed, did test 6 and project recovery re-opened the script from test 5 instead of the expected script from test 6

GuiMacielPereira, 2024-05-15:
When following test 8 we closed mantid normally but when we re-started it we still got the project recovery dialog

GuiMacielPereira, 2024-05-15:
Tried replicating running test 5 and then 6 but everything worked as expected this time. Total testing time around 30 mins.

closed 2024-05-15 by GuiMacielPereira