mantidproject/mantid #37358: Error from clicking on plot settings in a plot

GuiMacielPereira, 2024-05-14:
**Original reporter:** James Cresswell Email: [wbxl41@durham.ac.uk](mailto:wbxl41@durham.ac.uk)

**Describe the bug**
Opening the plot settings dialog in a plot caused an error:
```
File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/workbench/plotting/figuremanager.py", line 364, in launch_plot_options
    self.plot_options_dialog = PlotConfigDialogPresenter(self.canvas.figure, parent=self.window)
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/plotconfigdialog/presenter.py", line 45, in __init__
    curves_tab = CurvesTabWidgetPresenter(self.fig, parent_view=self.view, parent_presenter=self, legend_tab=legend_tab)
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 38, in __init__
    self.update_view()
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 282, in update_view
    curve_props = CurveProperties.from_curve(self.get_current_curve())
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/plotconfigdialog/curvestabwidget/presenter.py", line 92, in get_current_curve
    name = self.view.get_current_curve_name()
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/widgets/plotconfigdialog/curvestabwidget/view.py", line 78, in get_current_curve_name
    return self.select_curve_list.currentItem().text()
AttributeError: 'NoneType' object has no attribute 'text'
        Using: workbench 6.9.20240425.1854 on Rocky Linux 8.9 (Green Obsidian)
```

**To Reproduce**
I asked the user for the data he used and he sent me this folder: 
[MantidTroubleshooting.zip](https://github.com/mantidproject/mantid/files/15307167/MantidTroubleshooting.zip)

The script generated from the faulty plot was the following:
```python
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

import matplotlib.pyplot as plt
from mantid.plots.utility import MantidAxType
from mantid.api import AnalysisDataService as ADS

IvsQ_74306_74307_74308 = ADS.retrieve('IvsQ_74306_74307_74308')
IvsQ_74374_74375_74376 = ADS.retrieve('IvsQ_74374_74375_74376')
IvsQ_74324_74325_74326 = ADS.retrieve('IvsQ_74324_74325_74326')
IvsQ_74388_74389 = ADS.retrieve('IvsQ_74388_74389')
IvsQ_74342_74343_74344 = ADS.retrieve('IvsQ_74342_74343_74344')

fig, axes = plt.subplots(edgecolor='#ffffff', num='S12 MS024', subplot_kw={'projection': 'mantid'})
axes.errorbar(IvsQ_74342_74343_74344, capsize=1.0, color='#1f77b4', elinewidth=1.0, label='IvsQ_74342_74343_74344: spec 330', linewidth=1.0, marker='.', markersize=4.0, wkspIndex=0)
axes.errorbar(IvsQ_74324_74325_74326, capsize=1.0, color='#ff7f0e', elinewidth=1.0, label='IvsQ_74324_74325_74326: spec 330', linewidth=1.0, marker='.', markersize=4.0, wkspIndex=0)
axes.errorbar(IvsQ_74306_74307_74308, capsize=1.0, color='#2ca02c', elinewidth=1.0, label='IvsQ_74306_74307_74308: spec 330', linewidth=1.0, marker='.', markersize=4.0, wkspIndex=0)
axes.errorbar(IvsQ_74388_74389, capsize=1.0, color='#e377c2', elinewidth=1.0, label='S12 Si PET MS024 Rinse Lipid Air IvsQ_74388_74389: spec 330', linewidth=1.0, marker='.', markersize=4.0, wkspIndex=0)
axes.errorbar(IvsQ_74374_74375_74376, capsize=1.0, color='#d62728', elinewidth=1.0, label='IvsQ_74374_74375_74376: spec 330', linewidth=1.0, marker='.', markersize=4.0, wkspIndex=0)
axes.minorticks_on()
axes.tick_params(axis='x', which='minor', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 0, 'tickdir': 'out', 'width': 1})
axes.tick_params(axis='y', which='minor', **{'gridOn': False, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 0, 'tickdir': 'out', 'width': 1})
axes.tick_params(axis='x', which='major', **{'gridOn': True, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
axes.tick_params(axis='y', which='major', **{'gridOn': True, 'tick1On': True, 'tick2On': False, 'label1On': True, 'label2On': False, 'size': 6, 'tickdir': 'out', 'width': 1})
axes.set_title('S12 MS024')
axes.set_xlabel('q ($\\AA^{-1}$)')
axes.set_ylabel('Reflectivity')
axes.set_xlim([0.0079686, 0.39723])
axes.set_ylim([0.0, 3.4784])
axes.set_xscale('log')
axes.set_yscale('log')
legend = axes.legend(fontsize=8.0).set_draggable(True).legend

fig.show()
# Use plt.show() if running the script outside of Workbench
#plt.show()
# Scripting Plots in Mantid:
# https://docs.mantidproject.org/tutorials/python_in_mantid/plotting/02_scripting_plots.html
``` 

I tried replicating this but the plot options opened fine.

**Expected behavior**
No error.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Rocky Linux 8.9 (Green Obsidian)
 - Mantid Version 6.9.20240425.1854

**Additional context**
@jhaigh0 suggested that the function is changed to:
```
current_curve = self.select_curve_list.currentItem()
if current_curve:
    return current_curve.text()
return ""
```

sf1919, 2024-05-14:
Is this happening in earlier versions of Mantid?

GuiMacielPereira, 2024-05-14:
I tested it and it's not present in 6.8, it appeared in 6.9

sf1919, 2024-05-14:
In that case I think we'd best sort it out as a bugfix for v6.10 as it looks pretty small.

jhaigh0, 2024-05-15:
I recognise that line of code as something I had to change with the mpl update, so probably introduced the problem there.

closed 2024-05-20 by GuiMacielPereira