mantidproject/mantid #37369: Sliceviewer-Zooming in outside the extent of the data does not change Log colorscale option

warunawickramasingha, 2024-05-15:
While following the testing instructions for sliceviewer: https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id4
at step 5 when Zoom in to a region outside the extent of the data, the log colorscale was still enabled as below. But the test instructions says otherwise.
[image]

**Describe the bug**

**To Reproduce**

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

RichardWaiteSTFC, 2024-05-17:
I think this just needs updating in the manual testing docs - thanks!

sf1919, 2024-05-17:
@RichardWaiteSTFC maybe we can close this one and open a separate issue to update the testing docs as I think there is another issue that also could be solved by updating the docs?

RichardWaiteSTFC, 2024-05-17:
> @RichardWaiteSTFC maybe we can close this one and open a separate issue to update the testing docs as I think there is another issue that also could be solved by updating the docs?

Will do just want to check again the behaviour of the cursor info table described in issue 2 of the description

RichardWaiteSTFC, 2024-05-17:
Closing as all doc updates will be addressed in #37390

closed 2024-05-17 by RichardWaiteSTFC