mantidproject/mantid #37371: Sliceviewer-update in slicepoint does not update cursor info tabale

warunawickramasingha, 2024-05-15:
**Describe the bug**
When following the sliceviewer test instructions as listed : https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id5
at step 5, when the slicepoint is changed along one of the non-viewed axis, the slicepoint is not updated in the cursor info table

**To Reproduce**

1. Follow test instructions here: https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id5
2. at step 5 Untick track cursor
3. click on the plot
4. change slicepoint on one of non-viewed axis. But the values in the table do not change.

**Expected**:
the slicepoint is updated in the cursor info table.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

jclarkeSTFC, 2024-05-16:
This is also in 6.9.1

RichardWaiteSTFC, 2024-05-17:
We would not expect the values in the cursor info table to change if Track Cursor is unchecked - the values should update if you click on the colorfill (2D) plot

RichardWaiteSTFC, 2024-05-17:
Closing as all doc updates will be addressed in #37390

closed 2024-05-17 by RichardWaiteSTFC