mantidproject/mantid #37372: Live data plot doesn't adjust axes

jclarkeSTFC, 2024-05-15:
**Describe the bug**

If looking at a plot of live data, then the axes don't rescale as the line are redrawn, so the lines can wonder away until they're not visible any more.

**To Reproduce**

Live data manual testing [instructions](https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html), then follow one of the two `ADARA_FakeEvent` sections at the bottom of the page.

**Platform/Version (please complete the following information):**
 - OS: macOS (but I bet this happens on all operating systems)
 - OS Version: 11.5.2
 - Mantid Version v6.9.20240514.1920

**Additional context**

Found during manual testing for 6.10 release, see #37333.

jclarkeSTFC, 2024-05-16:
This also happens in 6.9.1

jhaigh0, 2024-06-06:
Weirdly if you click home a second time it then stops following the data permanently