mantidproject/mantid #37382: Ability to provide an output base name on Indirect/Inelastic interfaces

robertapplin, 2024-05-16:
**Blocked by #37381 **

Description
-------
Sanghamitra has mentioned it is frustrating when running a tab multiple times, the output workspaces in the ADS can be overriden. This happens when the loaded dataset remains the same, but the user might want to change the integrations range (on Elwin), so that they can compare the two outputs. She suggested we change the name of the workspace if the integration range is different, but this is not feasible for two reasons:
- The workspace name will start to get really long and hard to read
- Users will start asking for more data to be put into the workspace name to differentiate between workspaces

We therefore want to add the ability to specify an output base name somewhere within the Run Widget so that users do not override workspaces already output in the ADS. This output name should only be for the start of the output name, as we will need to make sure the end has the appropriate suffix such as `_red`. 

One idea for this output base name is to have a Dialog box to appear after clicking "Run" if the default output name is already stored in the ADS. It can ask the user if they want to choose a new name, or if they want to continue and override the existing workspace.

Acceptance Criteria
--------
- [ ] Implemented on Elwin tab first - so we can do some testing and get feedback from users

adriazalvarez, 2024-05-16:
Is it the same as this? #37201

robertapplin, 2024-05-16:
> Is it the same as this? #37201

Ah yes it is, thanks!

closed 2024-05-16 by robertapplin