mantidproject/mantid #37384: Update Engineering Diffraction manual testing instructions

rbauststfc, 2024-05-16:
Here are some suggested updates to the manual testing instructions for the Engineering Diffraction interface following manual testing for version 6.10:

- [ ] It would be good to introduce a comment at the top of the manual testing instructions warning about the potential difficulty with loading ENGINX files from the archive and how to get around this. This should also include an instruction for how to enter the range of runs specified in Test 4.
- [ ] For Test 5, in the instructions it's not clear that the settings you're asked to test are the browse filters for filtering the files that are available to select from the file browser dialog.
- [ ] In Test 9, it's not clear from the instructions where the primary log combo box is, as it doesn't seem to have this label on the interface.

sf1919, 2024-06-10:
@warunawickramasingha can this now be closed?

warunawickramasingha, 2024-06-10:
Closing this issue since all the above suggestions were captured in #37482.

closed 2024-06-10 by warunawickramasingha