mantidproject/mantid #37401: Memory leak when time slicing in ISIS Reflectometry GUI

rbauststfc, 2024-05-20:
**Original reporter:** Max at ISIS

**Describe the bug**

We've had reports of Mantid crashing due to memory use when users are auto-processing INTER experiment data while time slicing. I've done a bit of investigating and it looks to me like there is a memory leak somewhere when we time slice in the GUI.

INTER data has become larger since they got their new detector, so their workspaces do fill up the memory more when processing a large number of runs. I found that clearing the workspace list and re-processing didn't cause the memory to keep building up if not time slicing, suggesting there isn't a memory leak under normal use. However the memory does seem to build up when time slicing is enabled.

**To Reproduce**

1) Go to the `Interfaces` -> `Reflectometry` -> `ISIS Reflectometry`.
1) On the Runs tab, in the first row of the table on the right hand side enter run `77064` and angle `0.8`.
1) On the Event Handling tab, under the "Uniform slicing" section select the `(sec) slices` radio box and enter `5` to slice the data into 5 second slices.
1) On the Experiment Settings tab enter the following into the first row of the Settings Lookup table:
   - ROI: `3-258`
   - ROI Detector IDS: `3001-3256,4001-4256,5001-5256,6001-6256,7001-7256,8001-8256,9001-9256,10001-10256`
1) Back on the Runs tab, click the process button. This should complete successfully and create a set of workspace groups. Take a note of the memory usage at this stage.
1) Clear the workspace list using the button in Mantid Workbench. The memory use won't go down, but I don't think this in itself necessarily indicates a memory leak, as it can just be that the system hasn't retrieved the memory yet. To demonstrate the leak, in the Reflectometry GUI click to process the row again. The memory usage should continue to build up.

**Expected behavior**

On second processing of the data, the memory should not build up.

**Platform/Version (please complete the following information):**
 - Mantid Version - seems to be a problem introduced since version 6.9.1

rbauststfc, 2024-05-20:
@cailafinn would you be happy to double check this on IDAaaS and check if you see the same behaviour on your Linux machine (and maybe Mac)? I would recommend using the first set of testing instructions (i.e. the time slicing example) - I'm still checking if the script I've provided for the second way of causing the memory to fill up eventually leads all the way to a crash.

cailafinn, 2024-05-21:
Reproduced on Ubuntu 22.04. Took two repeats of clicking `Process` to crash on a system with 32GiB of RAM. When the ADS was cleared, memory usage.

### Breakdown:
- Start:
   - Mantid: 0.5GiB
   - System: 5.2GiB
- 1st Process:
   - Mantid: 15.8GiB
   - System: 20.4GiB
- ADS Clear
  - Mantid: 13.3GiB
  - System: 18.1GiB
- 2nd Process
   - Usage rose over 5-7 seconds before reaching the limits of the system until workbench froze and was crashed by the OOM Killer.

rbauststfc, 2024-05-21:
As Caila has suggested, we should see if [this PR](https://github.com/mantidproject/mantid/pull/37405) to address issues with jemalloc has any impact on this issue. We've not made any changes to the Reflectometry code since 6.9.1 that seem like they should be causing this, and it isn't happening on Windows, so I'll test on IDAaaS when that PR has gone in and see where we are then.

rbauststfc, 2024-05-23:
This doesn't appear to have been helped by the jemalloc change. We're still waiting on results from a Valgrind scan. In the meantime I've done some more testing on IDAaaS and have found this alternative way to replicate the behaviour:

1) Load data using `LoadEventNexus` with the following settings:
    - Filename = `INTER77064`
    - LoadMonitors = `True`
    - MonitorsLoadOnly = `Histogram`
2) From the algorithm dialog, run `ReflectometrySliceEventWorkspace` with the following settings:
    - InputWorkspace = name of loaded workspace
    - MonitorWorkspace = name of loaded monitor workspace
    - UseNewFilterAlgorithm = `True` (note the memory builds up regardless of how this is set)
    - TimeInterval = `5`

It should show the same behaviour as the Reflectometry GUI. If `ReflectometrySliceEventWorkspace` is called repeatedly via a script, though, then the memory doesn't seem to build up in the same way, it seems to be necessary to call it from a GUI.

rbauststfc, 2024-05-23:
Depending on the results of the Valgrind scan, if further investigation is needed it will need to be done by someone with access to a Linux machine as I still can't replicate on Windows.

rbauststfc, 2024-05-24:
I've done a bit more testing on IDAaaS and this is starting to look like a more general problem. I can replicate it using the `LoadISISNexus` algorithm to load a WISH dataset.

I've found that if you run the following script then it _**doesn't**_ build up the memory:

```
for i in range(3):
    LoadISISNexus(Filename="WISH57086", LoadMonitors="Include", OutputWorkspace="test")
    AnalysisDataService.clear()
```

However if you do the above in any of the following ways then the memory does build up:

- Run `LoadISISNexus` via the algorithm dialog and clear the ADS using the Clear button.
- Run `LoadISISNexus` via the algorithm dialog and clear the ADS by running `AnalysisDataService.clear()` in the script editor.
- Run `LoadISISNexus` via the script editor and clear the ADS using the Clear button.

warunawickramasingha, 2024-05-24:
The steps were followed in IDAaaS using `Valgrind` memory analyser tool to check for any memory leaks. The main observation was that there was a discrepancy in the amount of memory reported as leaked by Valgrind and the memory usage reported by the Mantid workbench as shown below. As can be seen, there has **not** been such a significant memory leak in this case.

- Valgrind summary ->

[image]

- System memory usage reported by Mantid at the end of test step 6 ->

[image]

cailafinn, 2024-05-28:
One billion permutations later, here are my results:

## Testing on Ubuntu 22.04. 32GiB RAM.

#### Run Script Once: Loop Loading Call
```python
for i in range(15):
    AnalysisDataService.clear()
    LoadISISNexus(Filename="WISH57086", LoadMonitors="Include", OutputWorkspace="test")
```
- Does not crash the program, even if the same load and clear function is performed multiple times within the script.
#### Run Script Multiple Times: Single Loading Call
```python
# Repeatedly ran 3 times.
AnalysisDataService.clear()
LoadISISNexus(Filename="WISH57086", LoadMonitors="Include", OutputWorkspace="test")
```
- With each re-run, the memory usage was seen to drop by around 2.3GiB when the ADS was cleared. 
- Memory usage then rose again. This could only be repeated 3 times before Mantid took up too much memory and was killed by the OoM Killer. 

| Version | Start | 1st Run | Clear | 2nd Run | Clear | 3rd Run |
| ------- | ----- | ------- | ----- | ------- | ----- | ------- |
| Nightly | 0.3   | 14.8    | 13.1  | 26.2    | 23.9  | CRASH   |
| 6.9.1   | 0.3   | 14.8    | 13.4  | 26.2    | 23.8  | CRASH   |
- Behaviour appears to be the same.
#### Run Script Multiple Times: Swap Clear Order
```python
# Repeatedly ran 4 times.
LoadISISNexus(Filename="WISH57086", LoadMonitors="Include", OutputWorkspace="test")
AnalysisDataService.clear()
```

| Version | Start | 1st Run | Clear | 2nd Run | Clear | 3rd Run | Clear | 4th Run | Clear | 5th Run |
| ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- |
| Nightly | 0.5   | 14.8    | 12.3  | 26.2    | 23.6  | 25.8    | 23.8  | CRASH   | N/A   | N/A     |
| 6.9.1   | 0.4   | 14.8    | 12.3  | 25.7    | 23.7  | 25.9    | 23.9  | 25.8    | 24.1  | CRASH   |
- Behaviour is similar. This way around was a bit more variable for some reason, but I don't think this one is specifically a regression. 
#### No Script
```python
# Click Load and load WISH57086 from the algorithm dialog
# Clear the ADS using the clear button. 
```

| Version | Start | 1st Run | Clear | 2nd Run | Clear | 3rd Run | Clear | 4th Run | Clear | 5th Run |
| ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- |
| Nightly | 0.4   | 14.8    | 12.3  | 26.2    | 23.7  | CRASH   |       |         |       |         |
| 6.9.1   | 0.4   | 14.8    | 12.3  | 14.5    | 12.5  | 14.6    | 12.7  | 14.7    | 12.9  | 14.7    |

#### Load with Algorithm Dialog, Clear From Script

```python
# Click Load and load WISH57086 from the algorithm dialog
# Run below from a script
AnalysisDataService.clear()
```

| Version | Start | 1st Run | Clear | 2nd Run | Clear | 3rd Run | Clear | 4th Run | Clear | 5th Run |
| ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- |
| Nightly | 0.5   | 14.8    | 12.3  | 26.2    | 24.9  | CRASH   |       |         |       |         |
| 6.9.1   | 0.5   | 14.7    | 12.3  | 14.8    | 12.5  | 14.7    | 13.1  | 14.5    | 13.2  | 14.7    |
- Again, matching the 6.9.1 scripted behaviour on this one. 

#### Load with a script, Clear from the Button
```python
# Run the below script to load WISH57086
# Then clear using the button on the workbench GUI.
LoadISISNexus(Filename="WISH57086", LoadMonitors="Include", OutputWorkspace="test")
```

| Version | Start | 1st Run | Clear | 2nd Run | Clear | 3rd Run | Clear | 4th Run | Clear | 5th Run |
| ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- |
| Nightly | 0.5   | 14.7    | 12.3  | 26.2    | 23.7  | CRASH   |       |         |       |         |
| 6.9.1   | 0.4   | 14.6    | 13.4  | 26.2    | 23.7  | CRASH   |       |         |       |         |
- Okay, so the total of all of this seems to tell us is that something has changed when *loading* from the algorithm dialog. Crashes occur on both versions when loading from scripts, regardless of what is done with the ADS clear. The only thing that stops the crashes from occurring in 6.9.1 is using the algorithm dialog to load the data.

cailafinn, 2024-05-28:
## Testing on macOS 14.4.1 (Intel). 32GB RAM.
#### No Script
```python
# Click Load and load WISH57086 from the algorithm dialog
# Clear the ADS using the clear button. 
```

| Version | Start | 1st Run | Clear | 2nd Run | Clear | 3rd Run | Clear | 4th Run | Clear | 5th Run |
| ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- | ----- | ------- |
| Nightly | 0.4   | 5.5     | 2.1   | 7.4     | 2.9   | 7.5     | 6.2   | 10.0    | 3.3   | 12.2    |
| 6.7     | 0.5   | 12.5    | 2.5   | 12.5    | 3.1   | 12.5    | 3.3   | 12.6    | etc.  |         |
- Despite the workspace appearing in the "Workspaces" list as 20GB, it didn't appear to take all that up as RAM. I wonder if there's something different about how macOS handles the memory for large files?
- Regardless, there doesn't seem to be an issue on macOS. This seems to be specifically for linux.

cailafinn, 2024-05-28:
`workbench` entry point in both cases. Always using the version from conda.

rbauststfc, 2024-05-28:
We've noticed that the version of the Nightly from 24th May (which is the first to include the jemalloc pin) is looking much better on IDAaaS now, when started with the launcher from the Applications menu (FYI @sf1919). Tom has explained that launching the Nightly using the IDAaaS launcher sets up jemalloc, whereas using the workbench entry point does not use the script that sets up jemalloc.

sf1919, 2024-06-12:
With the further changes to IDAaaS last week how are things looking @thomashampson ?

closed 2026-08-20 by github-actions[bot]